zephyrproject-rtos/zephyr #70187: Add a feature to retrieve the previous error/hang stack trace for current boot to increase the debuggability

mingfungchang, 2024-03-13:
**Is your feature request related to a problem? Please describe.**
In my use case, when a MCU warm reset triggered by Watchdog occurred due to a MCU hang. UART logging is the only way for debugging. In order to increase debuggability for the MCU hang scenario, it would be great to have a feature to retrieve previous stack trace in current.    
**Describe the solution you'd like**
I would like a function so that It could print out a buffer which logs the  stack trace info for previous boot. so that this info could be used to help debuggability.

**Describe alternatives you've considered**

**Additional context**

nordicjm, 2024-03-14:
There is something similar for this for instrumentation in a PR though it has not been updated in quite some time: https://github.com/zephyrproject-rtos/zephyr/pull/57371

That could probably be used as a starting point

gromero, 2024-03-14:
@mingfungchang cc @nordicjm I've just rebased and updated the PR with the latest development done by Erik, which now allows trying the instrumentation subsystem on a VM. In your case it could be helpful to adjust the trigger and stopper function (maybe the fault handler will be the point to stop tracing?). Note that it's necessary to provide a .overlay specific to your board for the retention mem (take a look at sample src/main.c comments and for the overlay example samples/boards/*.overlay). Mind that Instrumentation subsystem it for tracing and profiling, not so much for debugging, but as Jamie said it could be a good starting pointer for you. In the instrumentation subsystem data is written to the memory and then babeltrace is used to decode the binary stream in the host side). Cheers.