zephyrproject-rtos/zephyr #70213: tests: can: drivers.can.shell and net.socket.can fail

katgiadla, 2024-03-14:
**Describe the bug**
The tests `net.socket.can` and `drivers.can.shell` fail.

Observed for:
- `nrf52dk/nrf52832`
- `nrf52840dk/nrf52840`
- `nrf5340dk/nrf5340/cpuapp`
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf5340dk/nrf5340/cpunet`
- `nrf54l15pdk/nrf54l15/cpuapp`
- `nrf9160dk/nrf9160`
- `nrf9160dk/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/can/shell -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
In file included from zephyr/tests/drivers/can/shell/src/main.c:9:
zephyr/include/zephyr/drivers/can.h:825:15: error: 'can_get_min_bitrate' declared 'static' but never defined [-Werror=unused-function]
  825 | __syscall int can_get_min_bitrate(const struct device *dev, uint32_t *min_bitrate);
      |               ^~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
- Commit SHA or Version used: `v3.6.0-840-g3667526b832d`

henrikbrixandersen, 2024-03-14:
> In file included from zephyr/tests/drivers/can/shell/src/main.c:9:
> zephyr/include/zephyr/drivers/can.h:825:15: error: 'can_get_min_bitrate' declared 'static' but never defined [...]

I've just tried building locally here using `west build -b nrf52840dk/nrf52840 tests/drivers/can/shell/` - no issues. Did you clean your build directory? This is needed after the introduction of a new system call.

henrikbrixandersen, 2024-03-14:
```
$ ./scripts/twister -T tests/drivers/can/shell -p nrf52840dk/nrf52840 -v --inline-logs
ZEPHYR_BASE unset, using "/home/hebad/Projects/zephyrproject/zephyr"
Renaming output directory to /home/hebad/Projects/zephyrproject/zephyr/twister-out.2
INFO    - Using Ninja..
INFO    - Zephyr version: v3.6.0-776-g5988bc54b003
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/hebad/Projects/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 4
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 nrf52840dk/nrf52840       tests/drivers/can/shell/drivers.can.shell          PASSED (build)

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 21.30 seconds
INFO    - In total 41 test cases were executed, 0 skipped on 1 out of total 700 platforms (0.14%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/hebad/Projects/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/hebad/Projects/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/hebad/Projects/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

```

katgiadla, 2024-03-15:
the bug does not occurred again.

closed 2024-03-15 by katgiadla