zephyrproject-rtos/zephyr #70223: SPDX bom for twister builds

mkschreder, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
I would like to get spdx bom generated for builds that are done with twister using sample.yaml file (while we are at it, why not add another accepted name for this file such as "product.yaml" and make that standard way to configure "variants"?)

**Describe the solution you'd like**
A flag for twister that would automatically initialize the spdx bom before the build and then generate the bom after the build of each defined variant. 

**Describe alternatives you've considered**
The only alternative today seems to be to use west build directly.

**Additional context**
None

nordicjm, 2024-03-14:
> (while we are at it, why not add another accepted name for this file such as "product.yaml" and make that standard way to configure "variants"?)

No to this, sample.yaml is sufficient, we don't need more yaml files with the same thing

kartben, 2024-03-14:
> > (while we are at it, why not add another accepted name for this file such as "product.yaml" and make that standard way to configure "variants"?)
> 
> No to this, sample.yaml is sufficient, we don't need more yaml files with the same thing

The way I understand the original comment is a suggestion to use a more representative name for this file, not to *add* a new file. And simply looking at e.g `example-application/app/sample.yaml` makes me agree that there might be an opportunity to improve the naming of this file to better reflect its actual use

mkschreder, 2024-03-14:
Yes `example-application/app/sample.yaml` is a good example. sample.yaml makes it seem as though it is never intended to be used outside of samples directory. Another approach could be to merge sample.yaml and testcase.yaml into a single "twister.yaml". @nordicjm no to this as well from you?

nordicjm, 2024-03-14:
They can be combined, the schema for both is the same: https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/schemas/twister/testsuite-schema.yaml