zephyrproject-rtos/zephyr #70258: Build system chooses incorrect python interpreter

teburd, 2024-03-14:
**Describe the bug**
The python interpreter being found by cmake does not match the python interpreter in my path as would be expected.

**To Reproduce**
The only guaranteed method of reproduction I've found is...

* install nix (could be done on any linux machine, https://nixos.org/download/)
* `nix develop github:teburd/nix-flake-zephyr-dev-shell`
* `virtualenv env`
* `source env/bin/activate.fish`
* `pip install -r scripts/requirements.txt`
* `cd doc && make html-fast`

cmake shows 

```sh
-- Found Python3: /nix/store/gpvnsgf8zhqjjgk63ddx7cq2bv5vgcyh-python3-3.11.6-env/bin/python (found suitable version "3.11.6", minimum required is "3.8") found components: Interpreter 
```
Checking which python is in my path

``` sh
command -v python
/home/tburdick/z/env/bin/python
```

**Expected behavior**

The virtualenv python interpreter path is used and all the dependencies installed by pip are available.

**Additional Context**

When investigating this issue I found that the upstream cmake python module correctly finds the python on my path but the custom find logic in Zephyr's python.cmake does not.

Nix adjusts cmake environment variables to helpfully suggest where to find things, in this case I'd note that the CMAKE_PREFIX_PATH contains the found python path.

```sh
CMAKE_PREFIX_PATH=/nix/store/9k1raq5yprgzlgb72kj8pbi7d62khmvk-zephyr-sdk:/nix/store/1a7iwbpvvk3g9wznbagc6r4v6dvk0350-rimage:/nix/store/gpvnsgf8zhqjjgk63ddx7cq2bv5vgcyh-python3-3.11.6-env
```

Our python module uses find_program() which by default searches CMAKE_PREFIX_PATH according to the docs here https://cmake.org/cmake/help/latest/command/find_program.html

The upstream cmake python module *does not do this* as can be seen here https://gitlab.kitware.com/cmake/cmake/-/blob/master/Modules/FindPython/Support.cmake?ref_type=heads#L1845

The cmake python module as a whole has many very carefully (presumably) crafted cases to deal with the quirks in finding python. It is my opinion that we should re-use the upstream python module, but barring that, at least copy over much of the logic there.

teburd, 2024-03-14:
Related PR and discussion #65995

carlescufi, 2024-03-20:
@teburd does this only happen when building the doc or also when building samples?

teburd, 2024-03-20:
> @teburd does this only happen when building the doc or also when building samples?

Using twister or building docs I run into this.

I recall running into this with west at some point but it was unclear what the circumstances were.

using the upstream cmake python module in place of all the west/python logic we have has worked flawlessly for me.

nashif, 2024-03-20:
> Using twister or building docs I run into this.

please make this clear int he description, from the current details it seems like it is a doc building problem only.

aescolar, 2024-04-30:
@tejlmand or @nordicjm , @57300 can you please take a look at this?

tejlmand, 2024-05-02:
Removinmg blocker and downgrading the issue.

The issue is valid, but it's not a release blocker or a high prio issue because of:
- It only affect a `nix` package manager users, https://nixos.org/download/
- It only appears to affect doc builds
- It's possible to point to a specific Python by using `-DPython2_EXECUTABLE=/path/to/specific/python3` and thereby ensure a specific python is used.

tejlmand, 2024-05-02:
@teburd seems to be a nix self-inflicted issue because Nix sets `CMAKE_PREFIX_PATH=/nix/store/gpvnsgf8zhqjjgk63ddx7cq2bv5vgcyh-python3-3.11.6-env` which CMake's `find_program()` honors when looking up Python.

The reason it happens to work is because the upstream FindPython CMake module disregards the CMAKE_PREFIX_PATH, and thus you don't notice any issues when using Nix.
But I would actually blame the fact that nix only sets CMAKE_PREFIX_PATH on Nix, and not adjusting to take your `/home/tburdick/z/env/bin/python` into consideration is to blame on Nix.

However, I have now made a PR which adds support for Nix, but I still need to do additional testing to ensure that all existing issues reported on Python lookup are still fixed.

See #72232

PR #72232 (2024-05-24): cmake: find python path adjustment
Fixes: #70258

Specify NO_CMAKE_PATH to prevent CMake find_program from having a preference for searching CMAKE_PREFIX_PATH.

Some tool like NixOS / Nix package manager sets this variable to point inside the nix store, like this:
CMAKE_PREFIX_PATH=/nix/store/gpvnsgf8zhqjjgk63dd...-python3-3.11.6-env

thus causing `find_program()` to prefer this over the tool in the default path.

Support Nix environments by specifying NO_CMAKE_PATH and in addition support `VIRTUAL_ENV` environment setting for search path when set by the virtual environment.
commit 207da52e (2024-05-24): cmake: find python path adjustment

Fixes: #70258

Specify NO_CMAKE_PATH to prevent CMake find_program from having a
preference for searching CMAKE_PREFIX_PATH.

Some tool like NixOS / Nix package manager sets this variable
to point inside the nix store, like this:
CMAKE_PREFIX_PATH=/nix/store/gpvnsgf8zhqjjgk63dd...-python3-3.11.6-env

thus causing `find_program()` to prefer this over the tool in the
default path.

Support Nix environments by specifying NO_CMAKE_PATH and in addition
support `VIRTUAL_ENV` environment setting for search path when set by
the virtual environment.
closed 2024-05-24 by nashif