zephyrproject-rtos/zephyr #70281: Unable to build McuBoot with serial recovery enabled

AlbertPJoshi, 2024-03-15:
Build fails when mcuboot is configured with serial recovery.

`warning: ZCBOR (defined at modules/zcbor/Kconfig:7, modules/zcbor/Kconfig:7) has direct dependencies ZEPHYR_ZCBOR_MODULE || (ZEPHYR_ZCBOR_MODULE && 0) with value n, but is currently being y-selected by the following symbols:
 - MCUBOOT_SERIAL (defined at /home/devuser/Desktop/ESP32McuBootTest/my-workspace/bootloader/mcuboot/boot/zephyr/Kconfig.serial_recovery:7), with value y, direct dependencies !BOOT_FIRMWARE_LOADER (value: y), and select condition !BOOT_FIRMWARE_LOADER (value: y)

error: Aborting due to Kconfig warnings`

Followed intructions from [link:](https://docs.mcuboot.com/readme-zephyr.html)

henrikbrixandersen, 2024-03-17:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

AlbertPJoshi, 2024-03-18:
ok

closed 2024-03-18 by AlbertPJoshi

Eric-ERGS, 2024-07-20:
I'm having the same problem that is mentioned above trying to add mcuboot with a rpi-pico. I see this is close but I don't see a solution? Any updates or hits to this error.

DineshVaghela, 2024-10-18:
warning: MCUMGR (defined at subsys/mgmt/mcumgr/Kconfig:5) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: ZCBOR (=n).

Suggestion:

if you are using Zephyr >3.6.0 , upgrade to latest west.yml file and update the zephyr RTOS

the west.yml file should include ZCBOR Keyword between line number 20 to 50

After this above errors should resolve and build should work.