zephyrproject-rtos/zephyr #70283: samples: drivers: i2s echo: sample.drivers.i2s.echo fails on nrf5340dk/nrf5340/cpuapp/ns

katgiadla, 2024-03-15:
**Describe the bug**
During building, samples found the overlay:

```
 -- Found devicetree overlay: /home/sample_user/zephyrproject/zephyr/samples/drivers/i2s/echo/boards/nrf5340dk_nrf5340_cpuapp.overlay 
```

Even then the sample `sample.drivers.i2s.echo` reports timeout on `nrf5340dk/nrf5340/cpuapp/ns`.
The samples works after adding dedicated overlay for `ns` variant.

**To Reproduce bug**
Steps to reproduce the behavior:
1. have `nrf5340dk/nrf5340/cpuapp/ns` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/drivers/i2s/echo -p nrf5340dk/nrf5340/cpuapp/ns --device-testing --device-serial /dev/ttyACM1 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
INFO    - 4294/4294 nrf5340dk/nrf5340/cpuapp/ns samples/drivers/i2s/echo/sample.drivers.i2s.echo    FAILED Timeout (66.173s)
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: v3.6.0-382-g3203bd660c6c

PerMac, 2024-03-15:
@nordicjm @tejlmand this is the issue we've discussed and I understood that you've found the root cause. I am not sure who can be assigned to it...

nordicjm, 2024-04-10:
Fixed by #71149

closed 2024-04-10 by nordicjm