zephyrproject-rtos/zephyr #70285: Update CANopenNode to v4.0

viniciusmiguel, 2024-03-15:
**Is your enhancement proposal related to a problem? Please describe.**
The current version of CANOpenNode which zephyr references is very old.
The EDS Editor (v0.8) which can generate the object dictionary definition to the current CANOpenNode library lacks functionalities to proper create PDOs compared to the current released version (v4.2) it also has many bugs.
Using the old library also makes harder to develop since all the documentation is up to date with the current version being released by CANOpenNode.

**Describe the solution you'd like**
Zephyr should reference the new v4.0 CANOpenNode and update the module / sample project.

trupples, 2025-01-10:
Has there any work been done on this? (on a fork, I suppose, but I'm not sure how to look that up, my google-fu failed me)

If not, I'll try putting together a driver myself in the following weeks.

henrikbrixandersen, 2025-01-10:
> Has there any work been done on this? (on a fork, I suppose, but I'm not sure how to look that up, my google-fu failed me)
> 
> If not, I'll try putting together a driver myself in the following weeks.

I am working on a solution for Zephyr v4.1.0.

ckhardin, 2025-03-19:
> > Has there any work been done on this? (on a fork, I suppose, but I'm not sure how to look that up, my google-fu failed me)
> > If not, I'll try putting together a driver myself in the following weeks.
> 
> I am working on a solution for Zephyr v4.1.0.

Looking for a fork to try out CANopenNode v4.2 or similar, since we have a setup that generates the object dictionary using the latest CANopenEditor and trying to match that against the zephyr module.

trupples, 2025-03-19:
For the time being, what I used successfully were the NTURacingTeam forks:
https://github.com/NTURacingTeam/canopennode/
https://github.com/NTURacingTeam/zephyr/

In my use case, these only required a small number of tweaks, mainly to change `CO_Config.h` and to fix a race condition on NMT communication reset which would halt the whole application. I'd be happy to share patches, but the base implementation they have also works out of the box if you just want to try it out.

ckhardin, 2025-03-19:
> For the time being, what I used successfully were the NTURacingTeam forks: https://github.com/NTURacingTeam/canopennode/ https://github.com/NTURacingTeam/zephyr/
>
> In my use case, these only required [...]

Thanks we are experimenting with that setup - but, is there a notion of an ETA to get this bump (or equivalent) into the upstream zephyr proper? would it use the repo as a basis or another?

henrikbrixandersen, 2025-03-19:
> Thanks we are experimenting with that setup - but, is there a notion of an ETA to get this bump (or equivalent) into the upstream zephyr proper? would it use the repo as a basis or another?

As noted above, I am working on a solution on this. The plan was to have this ready for Zephyr v4.1, but unfortunately this has been delayed until v4.2. No need to begin a second effort for the same goal.

trupples, 2025-03-19:
Yep, I didn't mean to side step your effort. I am very excited for the future mainlined implementation, and am willing to contribute if given the chance, but for the time being, these forks were the next best thing, and I had deadlines that couldn't wait

ckhardin, 2025-03-19:
> As noted above, I am working on a solution on this. The plan was to have this ready for Zephyr v4.1, but unfortunately this has been delayed until v4.2. No need to begin a second effort for the same [...]

No worries - just trying to find out if there is a branch we can pull and preview since we are working against some deadlines as well and need to figure out what fits those timeframes.

dsseng, 2025-05-17:
> As noted above, I am working on a solution on this. The plan was to have this ready for Zephyr v4.1, but unfortunately this has been delayed until v4.2. No need to begin a second effort for the same [...]

@henrikbrixandersen Is there any TODO list or other way for a contributor like me to help you with the CANopenNode update? What's left to be done and could I help? Thanks!

Benni77, 2025-08-08:
Also interested in contributing and the version 4 update

KyleKotowick, 2025-10-07:
> After carefully considering this, I regret to inform that there are no current plans to update the in-tree CANopenNode support to v4.0. Please see [#94891 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/94891#issuecomment-3273903810)

The linked comment refers to supporting a common API for multiple third-party stacks, but it's unclear how that relates to an update to support the modern CANopenNode stack (v4).

Presently, it is very difficult to use the CANopenNode module in any way, since the CANopenNode's editor tool doesn't seem to support the older CANopenNode versions.

Is it not possible to update the Zephyr module integration to support CANopenNode v4, without the extra work for a common API as specified in #94891?