zephyrproject-rtos/zephyr #70317: llext: unexpected heap state assert by llext_unload k_heap_free call when extension uses threads and printk

laurenmurphyx64, 2024-03-15:
**Describe the bug**

I'm not 100% sure about this, but it seems like printk-ing more than a certain number of (???) chars across the main extension thread and any thread it creates causes an unexpected heap state assert to be triggered when llext is unloading the extension and freeing the heap memory.

```
I: Loaded extension threads_kernel_
Main printk 1
Main printk
M
Creating thread
Starting thread
Thread printk 1
Thread printk 2
Thread printk 3
Thread joined
D: freeing memory region 0
D: freeing memory region 2
D: freeing memory region 3
D: freeing memory region 4
ASSERTION FAIL [left_chunk(h, right_chunk(h, c)) == c] @ WEST_TOPDIR/zephyr/lib/heap/heap.c:183
	corrupted heap bounds (buffer overflow?) for memory at 0x20001a00
E: r0/a1:  0x00000004  r1/a2:  0x000000b7  r2/a3:  0x00000000
E: r3/a4:  0x000028ed r12/ip:  0x00000000 r14/lr:  0x000005ed
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00005746
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20000120 (test_load_unload_threads_kernel)
E: Halting system
```

 - I've observed failures on mps2/an521/cpu0, mps2/an385, mps3/an547, qemu_cortex_m3, mps3/an547/ns (seems unrelated since it can't even boot Zephyr, so I won't put it below), plus qemu_xtensa (different error)
 - I don't think this has ever worked, though the errors have changed with recent PRs to llext
 
**To Reproduce**
Steps to reproduce the behavior:
1. Checkout [this branch](https://github.com/laurenmurphyx64/zephyr/commit/71280eaf9e5aa3e1fda5858c2396b87e8b26d447)
2. Build and run
```
west build -p -b mps3/an547 tests/subsys/llext/simple -- -DCONFIG_ARM_MPU=n && west build -t run
```

Delete a single character from one of the printk statements in test_thread or the "Creating thread" and "Starting thread" printks in test_entry and re-run to watch it pass!

**Expected behavior**
Test completes with no issue.

**Impact**
Showstopper.

**Logs and console output**

```
$ west build -p -b mps2/an385 tests/subsys/llext/simple -- -DCONFIG_ARM_MPU=n && west build -t run
...
START - test_load_unload_threads_kernel_objects
D: Loading relocatable or shared elf
D: Finding ELF tables...
D: section 0 at 6cc: name 0, type 0, flags 0, addr 0, size 0
D: section 1 at 6f4: name 31, type 1, flags 6, addr 0, size 348
D: section 2 at 71c: name 27, type 9, flags 40, addr 0, size 136
D: section 3 at 744: name 37, type 1, flags 3, addr 0, size 0
D: section 4 at 76c: name 43, type 8, flags 3, addr 0, size 144
D: section 5 at 794: name 48, type 8, flags 3, addr 0, size 1024
D: section 6 at 7bc: name 138, type 1, flags 2, addr 0, size 163
D: section 7 at 7e4: name 150, type 1, flags 3, addr 0, size 8
D: section 8 at 80c: name 146, type 9, flags 40, addr 0, size 16
D: section 9 at 834: name 164, type 1, flags 30, addr 0, size 33
D: section 10 at 85c: name 173, type 1879048195, flags 0, addr 0, size 47
D: section 11 at 884: name 1, type 2, flags 0, addr 0, size 576
D: symtab at 11
[... 151 lines cut ...]
Creating thread
Starting thread
Thread printk 1
Thread printk 2
Thread printk 3
Thread joined
D: freeing memory region 0
D: freeing memory region 2
D: freeing memory region 3
D: freeing memory region 4
ASSERTION FAIL [left_chunk(h, right_chunk(h, c)) == c] @ WEST_TOPDIR/zephyr/lib/heap/heap.c:183
	corrupted heap bounds (buffer overflow?) for memory at 0x20001a00
E: r0/a1:  0x00000004  r1/a2:  0x000000b7  r2/a3:  0x00000000
E: r3/a4:  0x000028ed r12/ip:  0x00000000 r14/lr:  0x000005ed
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00005746
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20000120 (test_load_unload_threads_kernel)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 22.04)
 - Toolchain: Zephyr SDK
 - SHA: 20f747b

teburd, 2024-03-21:
> D: freeing memory region 4
> ASSERTION FAIL [left_chunk(h, right_chunk(h, c)) == c] @ WEST_TOPDIR/zephyr/lib/heap/heap.c:183
> 	corrupted heap bounds (buffer overflow?) for memory at 0x20001a00

This is troublesome and is indicating we've likely corrupted the heap in some way.

Possible reasons I can come up with...

* allocation pattern (with aligned/unaligned allocs) managed to catch some edge case that's been untested
* we really did muck with memory outside of the allocated bounds for some region here

The allocation pattern could be easily tested for by recreating it in a sample app. Overflowing the heap bounds for an allocation can be looked for using a debugger and hardware watch point on the heap size header from what I can tell

pillo79, 2024-03-26:
Made me worry as well, luckily it is a very subtle issue with this specific sample (and another thing to mention in the "llext docs"): **Zephyr macros that allocate stuff in specific sections do not (currently) work in llexts** - these sections are simply discarded. 
Yes, `K_THREAD_STACK_DEFINE`, I'm looking at you ;-)

Have not investigated the details but the corruption generates in `k_thread_start`; in my heavily patched example, it overflowed the BSS section and the corruption could be detected first thing in the thread. 

To test the theory, I moved the stack definition to Zephyr and `EXPORT_SYMBOL`'d it to the llext as a `void*` - all green. @laurenmurphyx64 can you test this on your end as well?

teburd, 2024-03-26:
> Made me worry as well, luckily it is a very subtle issue with this specific sample (and another thing to mention in the "llext docs"): **Zephyr macros that allocate stuff in specific sections do not (currently) work in llexts** - these sections are simply discarded:
> 
> ```
> ...
> D: section 5 name .noinit."WEST_TOPDIR/zephyr/tests/subsys/llext/simple/src/threads_kernel_objects_ext.c".0
> D: Not copied section .noinit."WEST_TOPDIR/zephyr/tests/subsys/llext/simple/src/threads_kernel_objects_ext.c".0
> ...
> D: unhandled symbol 18, name my_thread_stack, type tag 1, bind 0, sect 5
> ...
> ```
> 
> Yes, `K_THREAD_STACK_DEFINE`, I'm looking at you! 😉 Have not investigated the details of what happens then, but the corruption generates in `k_thread_start`; in my heavily patched example, it overflowed the BSS section and the corruption could be detected first thing in the thread.
> 
> To test the theory, I moved the stack definition to Zephyr and `EXPORT_SYMBOL`'d it to the llext as a `void*` - all green. @laurenmurphyx64 can you test this on your end as well?

Great find, thank you for taking a closer look @pillo79

laurenmurphyx64, 2024-03-26:
> To test the theory, I moved the stack definition to Zephyr and `EXPORT_SYMBOL`'d it to the llext as a `void*` - all green. @laurenmurphyx64 can you test this on your end as well?

Works for me, thanks so much!

pillo79, 2024-04-04:
Can this be closed as well in favor of #71013?

laurenmurphyx64, 2024-04-04:
> Can this be closed as well in favor of #71013?

@pillo79 I wouldn't mind, it's up to you and @teburd

teburd, 2024-04-05:
Yeah I think I need to rewrite it as a feature add to allow for maybe some of the static define macros to work in an extension perhaps.

laurenmurphyx64, 2024-04-05:
> Yeah I think I need to rewrite it as a feature add to allow for maybe some of the static define macros to work in an extension perhaps. To me this is no longer a bug but a notable missing feature.

Closing this as a bug, then

closed 2024-04-05 by laurenmurphyx64