zephyrproject-rtos/zephyr #70326: cmake: check_zephyr_package clutters namespace, prevents picolibc compilation

sigmundklaa, 2024-03-16:
**Describe the bug**
When using the `picolibc` module, the `options` variable of  `check_zephyr_package` [(link)](https://github.com/zephyrproject-rtos/zephyr/blob/d89e8052da79f9542a18cb14d53d5d2e00976ef1/share/zephyr-package/cmake/zephyr_package_search.cmake#L52) seems to clutter the CMake namespace, causing a collision with the `options` variable used in `picolibc`. This causes `CHECK_ONLY`, `SEARCH_PARENTS` and `VERSION_CHECK` to be appended as arguments to `gcc`, causing a compiler error.

By adding:
```cmake
message(WARNING ${options})
```
after the Zephyr include in [picolibc](https://github.com/picolibc/picolibc/blob/103849f26cf95838e7f198eae6537c4ccf33dd27/CMakeLists.txt#L61) this seems to be confirmed by the following output:
```
CMake Warning at modules/lib/picolibc/CMakeLists.txt:62 (message):
  CHECK_ONLYSEARCH_PARENTSVERSION_CHECK
```

Specifically, this is targeting `native_sim`, but I have not tested any other platform.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup a fresh Zephyr project following the [guide](https://docs.zephyrproject.org/latest/develop/getting_started/index.html)
2. Add `CONFIG_POSIX_API=y` and `CONFIG_PICOLIBC_USE_MODULE=y` to `prj.conf`
3. Add the below to `CMakeLists.txt`:
```cmake
cmake_minimum_required(VERSION 3.20)

find_package(Zephyr REQUIRED)
target_sources(app PRIVATE src/main.c)
```
4. Add a simple hello world in `src/main.c`
5. Compile and run with `cmake -DBOARD=native_sim -Bbuild . && cmake --build ./build --target all`

**Expected behavior**
Compilation without errors

**Impact**
Unable to compile

**Logs and console output**
The result from attempted compilation is below. Note that this is with the warning described above added to the `picolibc` module
```
cmake -DBOARD=native_sim -Bbuild . && cmake --build ./build --target all
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Loading Zephyr default modules (Freestanding).
-- Application: /home/sig/src/orbit/oec-dev/zephyr-clean
-- CMake version: 3.27.7
-- Found Python3: /home/sig/src/orbit/oec-dev/zephyr-clean/.venv/bin/python (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/sig/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/sig/src/orbit/oec-dev/zephyr-clean/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: native_sim, identifier: native
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.3 (/home/sig/.local/zephyr-sdk-0.16.3)
-- Found toolchain: host (gcc/ld)
-- Found Dtc: /home/sig/.local/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/sig/src/orbit/oec-dev/zephyr-clean/zephyr/boards/native/native_sim/native_sim.dts
-- Generated zephyr.dts: /home/sig/src/orbit/oec-dev/zephyr-clean/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/sig/src/orbit/oec-dev/zephyr-clean/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/sig/src/orbit/oec-dev/zephyr-clean/build/zephyr/dts.cmake
Parsing /home/sig/src/orbit/oec-dev/zephyr-clean/zephyr/Kconfig
Loaded configuration '/home/sig/src/orbit/oec-dev/zephyr-clean/zephyr/boards/native/native_sim/native_sim_defconfig'
Merged configuration '/home/sig/src/orbit/oec-dev/zephyr-clean/prj.conf'
Configuration saved to '/home/sig/src/orbit/oec-dev/zephyr-clean/build/zephyr/.config'
Kconfig header saved to '/home/sig/src/orbit/oec-dev/zephyr-clean/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /usr/bin/ld.bfd (found version "2.34") 
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Using ccache: /usr/bin/ccache
CMake Warning at modules/lib/picolibc/CMakeLists.txt:62 (message):
  CHECK_ONLYSEARCH_PARENTSVERSION_CHECK

-- Looking for xtensa/config/core-isa.h
-- Looking for xtensa/config/core-isa.h - not found
-- Configuring done (4.6s)
-- Generating done (0.1s)
-- Build files have been written to: /home/sig/src/orbit/oec-dev/zephyr-clean/build
[  0%] Building C object modules/picolibc/CMakeFiles/c.dir/newlib/libc/argz/argz_add.c.o
gcc: error: CHECK_ONLY: No such file or directory
gcc: error: SEARCH_PARENTS: No such file or directory
gcc: error: VERSION_CHECK: No such file or directory
make[2]: *** [modules/picolibc/CMakeFiles/c.dir/build.make:76: modules/picolibc/CMakeFiles/c.dir/newlib/libc/argz/argz_add.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4044: modules/picolibc/CMakeFiles/c.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```

**Environment:**
 - OS: Ubuntu 22.04
 - Toolchain: Zephyr SDK 0.16.3, host
 - Commit SHA 774aa3605f571b9f8fadc4e88d50bdd4222f7a2f

keith-packard, 2024-03-18:
maybe a fix could be done in the picolibc bits?

sigmundklaa, 2024-03-18:
Could do yes, but I would imagine setting a variable in the global scope called options could still cause problems for other modules and/or projects, no?

aescolar, 2024-03-19:
CC @tejlmand

aescolar, 2024-03-19:
@sigmundklaa I'm not able to reproduce the problem you describe.. Either by following your description or doing what you can see below.
(Would you be so kind as to tell if there is different in between what you do and just doing this below, as maybe I'm missing something in your description )
```
zephyr$ mkdir build; cd build
zephyr$ cmake -GNinja -DBOARD=native_sim ../samples/hello_world/ -DCONFIG_POSIX_API=y  && ninja
```
which works just fine.
(AFAIK setting `-DCONFIG_PICOLIBC_USE_MODULE=y` has no impact as it is already set indirectly by setting `CONFIG_POSIX_API`)

sigmundklaa, 2024-03-19:
@aescolar That's strange, I can compile just fine when using `samples/hello_world`, but when running with a custom hello world in `src/main.c`:
```c
#include <stdio.h>

int main(void)
{
    (void)printf("Hello World\n");

    return 0;
}
```
And a CMakeLists.txt with:
```cmake
cmake_minimum_required(VERSION 3.20)

find_package(Zephyr REQUIRED)
project(test)
target_sources(app PRIVATE src/main.c)
```
I am unable to compile (with `cmake -GNinja -DBOARD=native_sim .. -DCONFIG_POSIX_API=y && ninja`)

Did you try with a custom hello world, or just the sample?

aescolar, 2024-03-19:
Still I can't repro :( 
```
zephyr/repro/hello_world$ more `find . `

*** .: directory ***

::::::::::::::
./CMakeLists.txt
::::::::::::::
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.20.0)

#find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
find_package(Zephyr REQUIRED)
project(hello_world)

target_sources(app PRIVATE src/main.c)

*** ./src: directory ***

::::::::::::::
./src/main.c
::::::::::::::
/*
 * Copyright (c) 2012-2014 Wind River Systems, Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <stdio.h>

int main(void)
{
	(void) printf("Hello World\n");
	return 0;
}
::::::::::::::
./prj.conf
::::::::::::::
# nothing here

``` 
```
zephyr/repro/hello_world$ mkdir build && cd build
zephyr/repro/hello_world/build$ cmake -GNinja -DBOARD=native_sim .. -DCONFIG_POSIX_API=y && ninja 
```
Builds and runs just fine.
Maybe you have something funny in your environment or so?

sigmundklaa, 2024-03-26:
Hmm that is strange, the problem seems to have gone away after sourcing `zephyr-env.sh` so seems like there was something wonky in the environment. I don't how that is though, as it seems strange that the variable does not "leak" when the correct environment variables are set

aescolar, 2024-03-26:
Thanks @sigmundklaa I would then recommend we close this issue, and if later you find that there is some problem in Zephyr related to it reopen it.

closed 2024-04-28 by sigmundklaa