zephyrproject-rtos/zephyr #70332: rpi_pico: I2C master not working

ubieda, 2024-03-16:
**Describe the bug**
When using the RPi Pico in a sample with I2C enabled (e.g: Shell sample with I2C commands), the i2c commands are unexpectedly blocking, not toggling the I2C lines and as a result not working.

I've tried the exact same sample with other boards (e.g: stm32g0br1e and nrf52840) and I can confirm the behavior is as expected. Which leads me to believe that it's not a sample-issue, rather a board/target-driver issue.

Before filing this issue, I've reached out over discord to see if others have seen it. @petejohanson has confirmed seeing this issue as well. Link: https://discord.com/channels/720317445772017664/938474761405726800/1218422579417583666

**To Reproduce**
Steps to reproduce the behavior:
1. Build Shell sample with I2C commands enabled: `west build -b rpi_pico samples/subsys/shell/shell_module -- -DCONFIG_I2C=y -DCONFIG_I2C_SHELL=y`.
2. Flash board (e.g: With `west flash` using J-Link or copying `zephyr.uf2` when the device presents as MSD device over USB.
3. Wire the I2C bus to have pull-ups on I2C0 SDA (P4) and SCL (P5).
4. Connect over UART (TX: P0, RX: P1), and issue a shell command to scan the I2C0 bus:`i2c scan i2c@40044000`.
5. See how the I2C scan command freezes indefinitely.

[image]

**Expected behavior**
The I2C scan command shouldn't be blocking indefinitely, and just go over each I2C address, and list whether there's a device present or not (e.g: See screenshot for a nRF52840DK with a BME280 on 0x77).

[image]

krzysztofgawrys, 2024-04-30:
@ubieda 

This must be bug in i2c_dw driver that RPi Pico uses.
Dirty workaround, you can change

https://github.com/zephyrproject-rtos/zephyr/blob/b306e80e00d71196913228cc96cecb717d5357a1/drivers/i2c/i2c_shell.c#L79

to

`msgs[0].len = 1U;`

and i2c scanner will work.

devjoshi9031, 2024-05-12:
Hi @ubieda, did you try the suggestion from @krzysztofgawrys. I tried that but it only works when a device is connected to the I2C pins. If nothing is connected the physical pins, the output is same as what you described in the issue. Thanks! 

Also, I debugged this issue where we don't apply the workaround (i.e. `msg[0].len = 0U;`) and found that it is waiting forever to take the [semaphore](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_dw.c#L717) but the function [i2c_dw_transfer_complete](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_dw.c#L346) which gives the semaphore is never called from within i2c_dw_isr. Also, another thing to note is that even though it calls [i2c_dw_data_send] (https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_dw.c#L300), it is not writing the data to I2C port and just masking the TX_EMPTY interrupt because the length of the data to be sent is 0. I think this will not generate any further interrupts resulting in only idle thread being scheduled forever. 

I hope this makes sense and please let me know if you any questions. I am planning to spend some more time debugging this issue and hopefully provide a solution which can be merged in mainline. I'll keep updating this thread. Thanks!

ubieda, 2024-05-15:
@devjoshi9031 @krzysztofgawrys Thanks for your replies!

I have the following observations after some testing on my end:

1. I confirm what you've described is true: The `i2c_dw_transfer()` does not seem to handle correctly the use-case where `cur_msg->len = 0`, as such, the I2C transfer will not occur, while the semaphore will leave the calling thread waiting forever.
2. I also see the I2C scan command is held indefinitely when the bus does not have pull-ups (nothing wired to it), which could be interpreted as bus contention. IMO - there should be a way to fail-safe even in this case.

To address these, I'd suggest the following:
1. Change `i2c_dw.c` to handle the use-case where the msg-len is zero.
2. Add a fail-safe scheme on `i2c_dw.c` to timeout after a waiting period when there's bus contention and return a corresponding error code to the user (-EIO??).

LMK if any of you is planning to work on this. Otherwise I can take it as well!

devjoshi9031, 2024-05-15:
Hi @ubieda Thanks for suggestions and confirmation. I am willing to work on this one. 

1. I will take further looks into it this  weekend. From a brief glance at the code, it won't be a simple task. 
2. I was actually thinking the same thing. Instead of waiting K_FOREVER, we can wait for few milliseconds which will be configurable by KCONFIG options.

Thanks, I'll keep updating here.

ubieda, 2024-05-15:
@devjoshi9031 Thanks for volunteering tackling this! I'll assign it to you.

devjoshi9031, 2024-05-24:
Hi @ubieda, hope you are doing good. After digging a lot into the datasheet and online, it seems like DW hardware does not support sending just the address and getting a NACK/ACK signal. For example, to support length = 0 transfer, [micropython ](https://github.com/micropython/micropython/blob/df419137821a07a382eb78ce528c700870a4351c/ports/rp2/machine_i2c.c#L158)switches to soft I2C implementation rather than using the HW for this same reason. Thus, for now I have added a workaround where if the message length is zero and the buffer to be sent/received is not NULL, the [i2c_dw_transfer](https://github.com/devjoshi9031/zephyr_forked/blob/5cbc3c6b4131f28d6795e291ed213164087809a6/drivers/i2c/i2c_dw.c#L681) function changes the length of the transfer to be 1. This unblocks the case where the HW does not even start a transfer. Although, now it would send an additional byte of data to the available peripherals. Additionally, I have also added a timeout in [k_sem_take](https://github.com/zephyrproject-rtos/zephyr/blob/8c8e2f6243f147953ce3fadb999c4e2a5faf4f5e/drivers/i2c/i2c_dw.c#L717) function call as we talked earlier today. 

I wanted to check if would be possible for you to try my code and see if it solves the issues for you?  I have tried on my end and seems like the changes are working as expected. Here is the [commit ](https://github.com/devjoshi9031/zephyr_forked/commit/5cbc3c6b4131f28d6795e291ed213164087809a6#diff-217f70dd8d9a72d53e57865e56135ab159e9a059b357dd10fed008708dbd4176) if you or someone would like to take a look. Please keep in mind that I will change the commit message to comply with the contribution guidelines. 

Please let me know if you have any questions/suggestions regarding the changes. Thanks!

ubieda, 2024-06-09:
Hi @devjoshi9031, apologies for the delay getting back to this.

I don't have a strong preference on the workaround. Inherently I'd try to search for alternatives but it may very well be unavoidable. 

Feel free to create a PR and link it to this issue and let's let the community review this proposal.

Thanks!

devjoshi9031, 2024-06-17:
Hi @ubieda, Sorry for the late reply. I'll create a PR this week and let's see how it goes with the maintainers/reviewers.

Thanks!

PR #75290 (2024-07-04): drivers i2c I2C_DW:Fix I2C scan example
Solves two similar issues listed below:

Issue 1: I2C scanner example for DesignWare hardware gets stuck indefenitely resulting in system hang up. This is because DW driver does not handle 0 byte transfer correctly which is the case for I2C scan example. Fixed it by overwriting the msg length to 1 if it is 0 and the buffer is not NULL.

Issue 2: Similarly, if the I2C pins are not pulled up (nothing connected to I2C pins), the DW hardware does not actually sending the data (assuming contention on the bus) hence not releasing the semaphore resulting in cal [... cut]
commit a9e6fa4b (2024-07-04): drivers: I2C_DW: Fix I2C scan example

Solves two identical issues listed below:

Issue 1: I2C scanner example for DesignWare hardware gets stuck
indefenitely resulting in system hang up.This is because DW I2C driver
does not handle 0 byte transfer correctly which is the case for I2C
scan example.

Fixed it by overwriting the msg length to 1 if it is 0 and the
buffer is not NULL.

Issue 2: Similarly, if the I2C pins are not pulled up (nothing connected
 to I2C pins), the DW hardware does not actually send the data
(assuming contention on the bus) hence not releasing the semaphore
resulting in  [... cut]
closed 2024-07-04 by aescolar