zephyrproject-rtos/zephyr #70333: Deep sleep(system off) is not working with LVGL and display driver

RuslanSpartak, 2024-03-16:
After wake up of system if (!device_is_ready(display_dev)) return false.
If we add return our display driver is not working.
But if we skip return statement all works as expected.
Not sure why little vgl or zephyr thinks that display driver is not ready.

RuslanSpartak, 2024-03-16:
Seems like maybe before going to power off mode we need to disable something in lvgl. Not usre but would be nice to know.

faxe1008, 2024-03-18:
Hi @RuslanSpartak,

Can you maybe share what display and SoC you are using, as well as the Zephyr version you are experiencing these issues on?
How exactly are you using the power management subsystem? Can you maybe provide your configuration or short MRE as well?

RuslanSpartak, 2024-03-18:
nRF52840 and SSD1306. SDK version 2.5.0. No power management.

RuslanSpartak, 2024-03-18:
About deep sleep I meant SYSTEM OFF mode on nRF52840. After so called wake up - display driver function device_is_ready(display_dev) return false. If not use return fucntion I see that lvgl is working and all is fine. But question is why display driver function device_is_ready(display_dev) return false?

aescolar, 2024-03-19:
@RuslanSpartak you seem to be reporting an issue in the Nordic SDK. Unless you are quite sure that the issue is present in upstream Zephyr and can provide an appropriate way of reproduce it in vanilla Zephyr I would request you to file a bug report in the Nordic SDK.

closed 2024-03-19 by RuslanSpartak