zephyrproject-rtos/zephyr #70344: Current Atmel SAM flash driver will not erase last page

adkinsjd, 2024-03-17:
**Describe the bug**
I have been trying to erase the end of flash on an Atmel Sam4s(8b) using the current sam flash driver. When you attempt to erase the last page the driver returns an EFAULT error despite successfully erasing the last page. I believe this is because the success condition at https://github.com/zephyrproject-rtos/zephyr/blob/2f6b41a694ee47dac5c431a81b7673026b0908c3/drivers/flash/flash_sam.c#L397 

cannot be met for the final page, as it requires a start_offset of the last page+1 to meet the condition. I am unsure of the best way to fix the issue, but currently in my application I have added the following check after the erase page function runs and it seems to be working:

```
/* Check if we've reached the end of pages to erase */                                          
if (info->start_offset + info->size >= erase_data->section_end) {                               
         /* Succeeded, stop iterating */                                                             
         LOG_INF("Marking erase_data succeeded as true.");                                                      
         erase_data->succeeded = true;                                                               
         return false;                                                                               
}          
```

Note this can't replace the current check condition or else the final page would never be erased.

**To Reproduce**
Run flash_erase on an atmel sam chip for the last page of flash.

**Expected behavior**
No error is reported.

**Impact**
I believe I have solved the issue in my working tree, but I don't feel knowledgeable enough about the flash architecture to submit a PR.

**Logs and console output**
```
2024-03-17 16:52:44.852627:	[00:01:09.945,000] <dbg> flash_sam: sam_flash_erase: Erasing sector from 0x46000 to 0x80000
2024-03-17 16:52:44.863496:	[00:01:09.945,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x46000 of size 0x1000
2024-03-17 16:52:44.874021:	[00:01:09.965,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x47000 of size 0x1000
2024-03-17 16:52:44.879790:	[00:01:09.985,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x48000 of size 0x1000
2024-03-17 16:52:44.890265:	[00:01:10.005,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x49000 of size 0x1000
2024-03-17 16:52:44.900877:	[00:01:10.030,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x4a000 of size 0x1000
2024-03-17 16:52:44.906467:	[00:01:10.050,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x4b000 of size 0x1000
2024-03-17 16:52:44.917442:	[00:01:10.070,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x4c000 of size 0x1000
2024-03-17 16:52:44.927895:	[00:01:10.090,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x4d000 of size 0x1000
2024-03-17 16:52:44.938566:	[00:01:10.116,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x4e000 of size 0x1000
2024-03-17 16:52:44.943953:	[00:01:10.136,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x4f000 of size 0x1000
2024-03-17 16:52:44.954727:	[00:01:10.156,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x50000 of size 0x1000
2024-03-17 16:52:44.965497:	[00:01:10.176,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x51000 of size 0x1000
2024-03-17 16:52:44.971133:	[00:01:10.195,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x52000 of size 0x1000
2024-03-17 16:52:44.981563:	[00:01:10.215,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x53000 of size 0x1000
2024-03-17 16:52:44.992396:	[00:01:10.235,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x54000 of size 0x1000
2024-03-17 16:52:44.997922:	[00:01:10.255,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x55000 of size 0x1000
2024-03-17 16:52:45.008955:	[00:01:10.275,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x56000 of size 0x1000
2024-03-17 16:52:45.019503:	[00:01:10.295,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x57000 of size 0x1000
2024-03-17 16:52:45.029884:	[00:01:10.315,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x58000 of size 0x1000
2024-03-17 16:52:45.035813:	[00:01:10.341,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x59000 of size 0x1000
2024-03-17 16:52:45.046110:	[00:01:10.361,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x5a000 of size 0x1000
2024-03-17 16:52:45.056767:	[00:01:10.381,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x5b000 of size 0x1000
2024-03-17 16:52:45.062654:	[00:01:10.400,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x5c000 of size 0x1000
2024-03-17 16:52:45.073025:	[00:01:10.420,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x5d000 of size 0x1000
2024-03-17 16:52:45.083698:	[00:01:10.440,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x5e000 of size 0x1000
2024-03-17 16:52:45.094327:	[00:01:10.460,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x5f000 of size 0x1000
2024-03-17 16:52:45.099948:	[00:01:10.480,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x60000 of size 0x1000
2024-03-17 16:52:45.110736:	[00:01:10.500,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x61000 of size 0x1000
2024-03-17 16:52:45.121211:	[00:01:10.520,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x62000 of size 0x1000
2024-03-17 16:52:45.127038:	[00:01:10.540,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x63000 of size 0x1000
2024-03-17 16:52:45.137798:	[00:01:10.559,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x64000 of size 0x1000
2024-03-17 16:52:45.148319:	[00:01:10.585,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x65000 of size 0x1000
2024-03-17 16:52:45.158831:	[00:01:10.605,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x66000 of size 0x1000
2024-03-17 16:52:45.164528:	[00:01:10.625,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x67000 of size 0x1000
2024-03-17 16:52:45.175105:	[00:01:10.651,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x68000 of size 0x1000
2024-03-17 16:52:45.185737:	[00:01:10.671,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x69000 of size 0x1000
2024-03-17 16:52:45.192730:	[00:01:10.691,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x6a000 of size 0x1000
2024-03-17 16:52:45.201943:	[00:01:10.716,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x6b000 of size 0x1000
2024-03-17 16:52:45.212577:	[00:01:10.736,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x6c000 of size 0x1000
2024-03-17 16:52:45.223228:	[00:01:10.756,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x6d000 of size 0x1000
2024-03-17 16:52:45.229074:	[00:01:10.776,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x6e000 of size 0x1000
2024-03-17 16:52:45.239540:	[00:01:10.795,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x6f000 of size 0x1000
2024-03-17 16:52:45.250233:	[00:01:10.815,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x70000 of size 0x1000
2024-03-17 16:52:45.255787:	[00:01:10.835,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x71000 of size 0x1000
2024-03-17 16:52:45.266485:	[00:01:10.855,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x72000 of size 0x1000
2024-03-17 16:52:45.277139:	[00:01:10.875,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x73000 of size 0x1000
2024-03-17 16:52:45.282555:	[00:01:10.895,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x74000 of size 0x1000
2024-03-17 16:52:45.293350:	[00:01:10.921,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x75000 of size 0x1000
2024-03-17 16:52:45.304075:	[00:01:10.941,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x76000 of size 0x1000
2024-03-17 16:52:45.314846:	[00:01:10.960,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x77000 of size 0x1000
2024-03-17 16:52:45.320146:	[00:01:10.980,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x78000 of size 0x1000
2024-03-17 16:52:45.331103:	[00:01:11.000,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x79000 of size 0x1000
2024-03-17 16:52:45.341622:	[00:01:11.020,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x7a000 of size 0x1000
[... cut ...]
2024-03-17 16:52:45.384813:	[00:01:11.126,000] <dbg> flash_sam: sam_flash_erase_page: Erasing page at 0x7f000 of size 0x1000
2024-03-17 16:52:45.395485:	[00:01:11.145,000] <err> firmwareupdate: Failed to erase flash: -14. Cannot proceed with update
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Zephyr SDK
 - Out of tree - but flash_sam.c and flash_page_layout.c (which invokes the function which produces the appropriate stop condition) are up to date with main.

nandojve, 2024-03-18:
Hi @adkinsjd ,

We really appreciate when community sends PR. Take a look at [contribute-guidelines](https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribute-guidelines) for more info about how to do it.

nandojve, 2024-05-19:
Hi @adkinsjd ,
Do you have plans to send the patch ?

nandojve, 2024-11-03:
Hi @bjarki-andreasen ,

Just checking. You made a lot of improvements on the SAM flash driver. I was wondering if you know about this issue.
Do you have any problems related with this?

bjarki-andreasen, 2024-11-03:
I do see the bug in the code, I never ran into this since I did not use the last section of flash in my project, I will post a PR with a fix :)

bjarki-andreasen, 2024-11-03:
@adkinsjd could you check out this PR and test it on your hardware? I no longer have hardware available to test with :)

bjarki-andreasen, 2024-11-03:
or @nandojve could you test it?

PR #80800 (2025-03-10): drivers: flash: sam: fix flash erase last page
The implementation for erasing pages in the flash_sam.c driver indicated a successful erase after exceeding the last page to be erased successfully. Since the last page has no proceeding page, the succeeded status was not set.

This fix switches around the status to be set as succeeded before erase begins, to have the succeeded status cleared if page unlock or erase fails.

fixes: #70344
PR #86879 (2025-03-13): [Backport v4.1-branch] drivers: flash: sam: fix flash erase last page
Backport 6f45f682c44b857f7cf9be8cbdfeb938194e8f21~2..6f45f682c44b857f7cf9be8cbdfeb938194e8f21 from #80800.

Fixes: #70344
closed 2025-03-10 by kartben