zephyrproject-rtos/zephyr #70353: Bluetooth: The work `struct bt_conn::tx_complete_work`  cannot be submitted due to the flag K_WORK_CANCELING_BIT of k_sys_work_q is set

lylezhu2012, 2024-03-18:
**Describe the bug**

After the hci_num_completed_packets is received, the pending conn_tx will be appeded to the queue `struct bt_conn::tx_complete`. And the work `struct bt_conn::tx_complete_work` will be submitted.
But the work `struct bt_conn::tx_complete_work`  cannot be submitted due to the flag K_WORK_CANCELING_BIT of k_sys_work_q is set. 
When the issue occurs,, the error code `-EBUSY` will be returned by `k_work_submit(&conn->tx_complete_work);` in function `hci_num_completed_packets` of file `hci_core.c`. While the error code is ignored by the function `k_work_submit(&conn->tx_complete_work);`.

Because the error code is ignored, this causes the Bluetooth Host stack to hang.

[image]
Form the debug log, there are three conn_tx 0x5659aa80,0x5659aa60,0x5659aa70 completed. 

[image]
After check all appended note of `struct bt_conn::tx_complete`, all of these node are appended.

[image]
Meanwhile, the BIT K_WORK_CANCELING_BIT of `falgs` is set in `k_sys_work_q`.

jori-nordic, 2024-03-18:
Is this for classic or LE?

lylezhu2012, 2024-03-18:
> Is this for classic or LE?

I found this issue in BR/EDR examples. However, I think this issue should not be unique to BE/EDR. From a code perspective, the function `hci_num_completed_packets` does not capture and handle all possible exceptions.

jhedberg, 2024-03-19:
@lylezhu2012 do you have a fix proposal?

lylezhu2012, 2024-03-20:
> @lylezhu2012 do you have a fix proposal?

After in-depth research, I found that the cause of the issue was inconsistent with my previous description. The work work `struct bt_conn::tx_complete_work` has been submitted, but it is not executed. The system work queue is in canceling state. The pending work is not executed normally.
The function `bt_conn_create_pdu_timeout` is called in system work queue thread context. It blocks the thread of `k_sys_work_q`. It make the system work queue to hang.
It seams the wrong usage. The function `bt_conn_create_pdu_timeout` cannot be called in context of the thread of `k_sys_work_q`.

closed 2024-03-20 by lylezhu2012