zephyrproject-rtos/zephyr #70360: scripts: ci: check_compliance fails custom projects file that contain DOS line endings

m-alperen-sener, 2024-03-18:
The problem:
We have projects files for Bluetooth Launch Studio and PTS application [here](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/bluetooth/qualification).  Compliance check fails on bls (launch studio project file) files because of line endings which is required by the launch studio, it is not possible to import the file back to launch studio if we change the line endings. 

Suggested solution:
Bypass compliance check for bls files.

henrikbrixandersen, 2024-03-18:
How is this an issue for the Zephyr Project? Sounds more like an issue for your downstream project/application?

m-alperen-sener, 2024-03-19:
More detail:

This [file](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bluetooth/qualification/Project_Zephyr_Bluetooth_Host.bls) is the project file that we use in Bluetooth Launch Studio to add/delete/update ICS for Zephyr Bluetooth Controller/Host/Mesh qualification. 

Bluetooth Launch Studio web app only accepts files with DOS line endings, so storing this particular file with DOS line ending is mandatory-ish.

The current compliance check is not possible to pass for this file.

PavelVPV, 2024-03-19:
This issue actually prevents this PR to be merged: https://github.com/zephyrproject-rtos/zephyr/pull/68975
Explanation: https://github.com/zephyrproject-rtos/zephyr/pull/68975#issuecomment-1996839934

henrikbrixandersen, 2024-03-19:
@m-alperen-sener, @PavelVPV Thank you for explaining this.

Thalley, 2024-08-28:
This is also an issue for certain Bluetooth configuration files that needs to have DOS line endings to work with testing software

nashif, 2024-08-28:
this was fixed in 420ffdd8a88

Thalley, 2024-08-28:
> this was fixed in [420ffdd](https://github.com/zephyrproject-rtos/zephyr/commit/420ffdd8a88ac0d06f6a7b677b4a2b8714f4374f)

@nashif 
Can you elaborate on how reducing the size of the error report helps with DOS line endings? Won't CI still fail?

nashif, 2024-08-28:
> @nashif
> Can you elaborate on how reducing the size of the error report helps with DOS line endings? Won't CI still fail?

CI will fail, of course, but this is not the point or problem. the problem was that CI was getting stuck with 10s of thousands of checkpatch violation and processing them one by one, now we do not do that and fail with showing the errors directly and without trying to inline them on github.

DOS line endings is still a rule violation and we need to fail on it and apply exceptions where applicable.

So I am not sure to be honest what the original enahncement is about? We always have false positives that we override. The commit above was for a similar issue, do not remember which PR it was, there the compliance checks was getting stuck because of the thousands of DOS line endings.

Thalley, 2024-08-28:
Thanks. I agree that the compliance not finishing was an issue, but this issue was specifically for DOS line endings, which is still an issue regardless. 

I do not consider https://github.com/zephyrproject-rtos/zephyr/pull/77283 a fix for this issue, and we should reopen, as the issue with DOS line ending still exist and ideally we would have some way of disregarding that check for specific files.

PavelVPV, 2024-09-14:
The issue still exists: https://github.com/zephyrproject-rtos/zephyr/pull/78400