zephyrproject-rtos/zephyr #70390: Compiling tests/drivers/can/api shows warning

talih0, 2024-03-18:
**Describe the bug**
When compiling CAN api test tests/drivers/can/api there's a warning message:

[image]

The issue only appears to happen when CONFIG_CAN_STATS=y

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b mimxrt1064_evk tests/drivers/can/api/

**Expected behavior**
No warning messages.

**Environment (please complete the following information):**
 - Linux
 - Toolchain Zephyr-SDK 0.16.5
 - Commit SHA d921659d6a4ffa9d47896c23f9292c238c183d19 (origin/main)

henrikbrixandersen, 2024-03-18:
I am unable to reproduce the above on current `main` (3fd545b05b45ec62a67b426ce809c804c86b8593):

```
$ west build -p -b mimxrt1064_evk tests/drivers/can/api/
-- west build: making build dir /home/brix/Projects/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/brix/Projects/zephyrproject/zephyr/tests/drivers/can/api
-- CMake version: 3.28.1
-- Found Python3: /home/brix/Projects/zephyrproject/venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/brix/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/brix/Projects/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: mimxrt1064_evk, identifier: mimxrt1064
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/opt/zephyr-sdk/0.16.5)
-- Found toolchain: zephyr 0.16.5 (/opt/zephyr-sdk/0.16.5)
-- Found Dtc: /opt/zephyr-sdk/0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/brix/Projects/zephyrproject/zephyr/boards/nxp/mimxrt1064_evk/mimxrt1064_evk.dts
-- Generated zephyr.dts: /home/brix/Projects/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/brix/Projects/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/brix/Projects/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/brix/Projects/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/brix/Projects/zephyrproject/zephyr/boards/nxp/mimxrt1064_evk/mimxrt1064_evk_defconfig'
Merged configuration '/home/brix/Projects/zephyrproject/zephyr/tests/drivers/can/api/prj.conf'
Configuration saved to '/home/brix/Projects/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/brix/Projects/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/zephyr-sdk/0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
Load components for MIMXRT1064:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_flexcan component is included.
driver_lpuart component is included.
driver_igpio component is included.
driver_cache_armv7_m7 component is included.
-- Using ccache: /usr/bin/ccache
-- Configuring done (9.8s)
-- Generating done (0.3s)
-- Build files have been written to: /home/brix/Projects/zephyrproject/zephyr/build
-- west build: building application
[  1%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[  2%] Built target parse_syscalls_target
[  3%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[  3%] Built target kobj_types_h_target
[  4%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  4%] Built target syscall_list_h_target
[  4%] Generating include/generated/driver-validation.h
[  4%] Built target driver_validation_h_target
[  4%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[  4%] Built target offsets
[  5%] Generating include/generated/offsets.h
[  5%] Built target offsets_h
[  5%] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/brix/Projects/zephyrproject/zephyr), build: v3.6.0-1180-g3fd545b05b45
[  5%] Built target version_h
[  5%] Built target zephyr_generated_headers
[  6%] Building C object CMakeFiles/app.dir/src/common.c.obj
[  6%] Building C object CMakeFiles/app.dir/src/classic.c.obj
[  7%] Building C object CMakeFiles/app.dir/src/transceiver.c.obj
[  7%] Building C object CMakeFiles/app.dir/src/utilities.c.obj
[  8%] Building C object CMakeFiles/app.dir/src/canfd.c.obj
[  9%] Building C object CMakeFiles/app.dir/src/stats.c.obj
[  9%] Linking C static library app/libapp.a
[... 186 lines cut ...]
[ 95%] Generating kobject_hash.c
[ 95%] Building C object zephyr/CMakeFiles/kobj_hash_output_lib.dir/kobject_hash.c.obj
[ 95%] Built target kobj_hash_output_lib
[ 96%] Generating kobject_hash_renamed.o
[ 96%] Built target kobj_hash_output_obj_renamed
[ 97%] Generating linker.cmd
[ 97%] Built target linker_zephyr_final_script_target
[ 98%] Generating isr_tables.c, isr_tables_vt.ld, isr_tables_swi.ld
[ 98%] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[100%] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[100%] Linking C executable zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      126296 B         4 MB      3.01%
             RAM:       15672 B        32 MB      0.05%
            ITCM:          0 GB       128 KB      0.00%
            DTCM:          0 GB       128 KB      0.00%
           OCRAM:          0 GB       256 KB      0.00%
          OCRAM2:          0 GB       512 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/brix/Projects/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: mimxrt1064_evk
[100%] Built target zephyr_final
[100%] Built target check_init_priorities
```

Do you have any custom changes, Kconfig options, environment variables, ... set?

talih0, 2024-03-18:
I had a default Kconfig option in my .west/config file. It should be reproducible with:

west build -p -b mimxrt1064_evk tests/drivers/can/api/ -- -DCONFIG_COMPILER_SAVE_TEMPS=y

henrikbrixandersen, 2024-03-18:
> I had a default Kconfig option in my .west/config file. It should be reproducible with:
> 
> 
> 
> west build -p -b mimxrt1064_evk tests/drivers/can/api/ -- -DCONFIG_COMPILER_SAVE_TEMPS=y

Okay. The issue is likely of the same nature as what was fixed in https://github.com/zephyrproject-rtos/zephyr/pull/54308, then.

henrikbrixandersen, 2024-03-19:
Same problem exists in `include/zephyr/drivers/i2c.h` when `CONFIG_I2C_STATS` is enabled.

henrikbrixandersen, 2024-03-19:
@talih0 Please try the proposed fix in https://github.com/zephyrproject-rtos/zephyr/pull/70439

PR #70439 (2024-04-05): include: zephyr: sys: util: add common is_null_no_warn() utility function and use it to squelch warning with CONFIG_COMPILER_SAVE_TEMPS=y
Some macros may need to check their arguments against NULL to support multiple use-cases, but NULL checks can generate warnings for a given use of those macros (where that particular argument can never be NULL).
    
This can happen if:
a) all macros are expanded (e.g. when using `CONFIG_COMPILER_SAVE_TEMPS=y`)
or
b) tracking of macro expansions are turned off (`-ftrack-macro-expansion=0`)
    
This warning can be circumvented by passing the argument to be c [... cut]
commit 2f136f21 (2024-04-05): drivers: can: use is_null_no_warn() check in generic macro

Use is_null_no_warn() for checking the init_fn against NULL in order to
avoid compiler warning when using CONFIG_COMPILER_SAVE_TEMPS=y.

Fixes: #70390
closed 2024-04-05 by carlescufi