zephyrproject-rtos/zephyr #70436: SPI clock is not working properly,

DinkarY, 2024-03-19:
Hello,

I am using nrf52840DK board.
I am trying to build an SPI interface with an external SPI sensor. For that, I made a little code that send one bytes of data and receive one byte of data but that is not working properly. then I checked an SPI pin signal and found that an SPI clock is not continues for both bytes. Find attached signal file.
[image]

I also attached the main.c file that i used.
[main.zip](https://github.com/zephyrproject-rtos/zephyr/files/14651335/main.zip)

please look into this and help me to resolve it.

Thanks,
Dinkar

htcrane, 2024-05-31:
I am seeing the same issue as well on a custom board I have with the NRF5340. I am using the `spi_spi_transceive_cb()` function. I notice when I have only one SPI buffer set active (the other being NULL like if I am just txing, the rx is NULL), the function works as expected. However, if both SPI buffer sets are active (neither rx nor tx are NULL), the receiving SPI buffer set never sends SPI clocks. 

**Notes**
- I am using the SPIM module as reflected in my device tree.
- The receive buffer internal values will be updated just, no clocks will be generated no matter the length of the rx buffer. The only way I can get the number of clocks to change is to increase the write buffer length. However this is not the expected behavior.
- I have attached a screenshot below of a 1 byte write (0x01) and a 1 byte read. See no clocks are generated for the read. I have stepped through the code and verified that all buffer sets are correctly passed. 

[image]

Tagging those who recently introduced changes to the SPIM driver: @anangl, @carlescufi, @jaz1-nordic, @awojasinski. Any thoughts?

anangl, 2024-06-10:
@DinkarY
> I am using nrf52840DK board. I am trying to build an SPI interface with an external SPI sensor. For that, I made a little code that send one bytes of data and receive one byte of data but that is [...]

The screenshot you provided shows a correct transaction performed for the code that you used:
```
	static uint8_t tx_buffer[2];
	static uint8_t rx_buffer[2];

	const struct spi_buf tx_buf = {
		.buf = tx_buffer,
		.len = 1	//sizeof(tx_buffer)
	};
	const struct spi_buf_set tx = {
		.buffers = &tx_buf,
		.count = 1
	};

	struct spi_buf rx_buf[] = {
		{.buf = NULL,
		.len = 1},
		{.buf = rx_buffer,
		.len = 1}
	};
	const struct spi_buf_set rx = {
		.buffers = rx_buf,
		.count = 2
	};

[..]
	
	int error = spi_transceive(spi_dev, &spi_cfg, &tx, &rx);
```
For such code, the `spi_nrfx_spim.c` driver needs to perform two transfers:
- TX one byte from `tx_buffer` (discard what was received)
- RX one byte and write it to `rx_buffer`.

And you see these two transfers. But since the CS line is kept active, it is still one transaction. The clock signal does not need to be continuous for this transaction to be correct. If you want the clock signal to be continuous, use:
```
	const struct spi_buf tx_buf = {
		.buf = tx_buffer,
		.len = 2
	};
```
and
```
	struct spi_buf rx_buf = {
		.buf = rx_buffer,
		.len = 2
	};
	const struct spi_buf_set rx = {
		.buffers = &rx_buf,
		.count = 1
	};
```
and discard `rx_buffer[0]`.

anangl, 2024-06-10:
@htcrane 
> I am seeing the same issue as well on a custom board I have with the NRF5340. I am using the `spi_transceive_cb()` function. I notice when I have only one SPI buffer set active (the other being NULL like if I am just txing, the rx is NULL), the function works as expected. However, if both SPI buffer sets are active (neither rx nor tx are NULL), the receiving SPI buffer set never sends SPI clocks.

There was a problem in the async transactions (the CS signal was deactivated too early) that was recently fixed by the following commit: 7080f0f83f9aea1052de28d8009c09279bee9c8c, but I'm not sure that was actually what you observed. Could you provide a piece of code that demonstrates the incorrect behavior?

> * I have attached a screenshot below of a 1 byte write (0x01) and a 1 byte read. See no clocks are generated for the read. I have stepped through the code and verified that all buffer sets are correctly passed.

If you use only one buffer for TX and only one for RX (in both `spi_buf_set` structures the `count` field is set to 1) and those are 1-byte buffers, there will be only 8 clocks (the buffers are processed in parallel), so I don't quite see anything wrong here.

carlescufi, 2024-07-10:
Closing due to inactivity from the reporter.

closed 2024-07-10 by carlescufi