zephyrproject-rtos/zephyr #70445: build failure of samples/drivers/spi_flash with  em_starterkit@2.2/emsk_em7d

FRASTM, 2024-03-19:
**Describe the bug**
Ci failure when building the samples/drivers/spi_flash for the target em_starterkit@2.2/emsk_em7d

`./samples/drivers/spi_flash/boards/em_starterkit.overlay devicetree error: /soc/spi@f0006000: undefined node label 'creg_gpio'
`

**To Reproduce**
Steps to reproduce the behavior:
1. em_starterkit@2.2/emsk_em7d
2. See error

**Expected behavior**
build successful and CI OK

**Impact**
CI failure on https://github.com/zephyrproject-rtos/zephyr/pull/70434

**Logs and console output**
```
-- Found devicetree overlay: ./zephyrproject/zephyr/samples/drivers/spi_flash/boards/em_starterkit.overlay
devicetree error: /soc/spi@f0006000: undefined node label 'creg_gpio'
-- In: ./zephyrproject/zephyr/build/zephyr, command: /usr/bin/python3;./zephyrproject/zephyr/scripts/dts/gen_defines.py;--dts;./zephyrproject/zephyr/build/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;./zephyrproject/zephyr/dts/bindings;--header-out;./zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;./zephyrproject/zephyr/build/zephyr/zephyr.dts.new;--edt-pickle-out;./zephyrproject/zephyr/build/zephyr/edt.pickle;--vendor-prefixes;./zephyrproject/zephyr/dts/bindings/vendor-prefixes.txt
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain zephyr-sdk-0.16.3
 - Commit SHA aeea1d10de8859d9368cbf07be0b9c3726c79a67

nordicjm, 2024-03-20:
Failing due to using overlay which should not apply to this variant. Other failures with the same issue:

- https://github.com/zephyrproject-rtos/zephyr/pull/69668

erwango, 2024-04-10:
@tejlmand It seems #71149 didn't fix current issue.

nordicjm, 2024-04-10:
> @tejlmand It seems #71149 didn't fix current issue.

How so?