zephyrproject-rtos/zephyr #70457: Inconsistency in userspace implementations

ceolin, 2024-03-19:
**Describe the bug**

When userspace is enabled using MMU in Zephyr, a thread has access to all other threads' stacks within the same memory domain. However, when using MPU, threads do not have access to other threads' stacks. 

When looking the usermode threat model, in the design goal section we can see:

 - A user thread will never by default have access to user thread stacks that are not members of the same memory domain.
    ...

 - A user thread may have read/write access to the stacks of other user threads in the same memory domain, depending on hardware.
   - On MPU systems, threads may only access their own stack buffer.
   - On MMU systems, threads may access any user thread stack in the same memory domain. Portable code should not assume this

But in several situations, we just state that Zephyr's userspace prevents a thread having access to other thread stack.

**Expected behavior**

Same userspace behavior independently if it is using MMU or MPU.

**Impact**

It depends on which behavior we decided. If we decide to isolate thread stacks for threads within the same memory domain
we will need to have page tables per thread or change the page table contents in context switch. If we decide to isolate thread stacks within a memory domain, we will need additional MPU changes in context switch.
 
**Additional context**

https://github.com/zephyrproject-rtos/zephyr/pull/67938#issuecomment-2007286475

andyross, 2024-03-19:
(Hah, crossed with discussion in the original PR.  Re-pasting here.)

OK, so here's how I think this mess happened: **It's not a framework security bug**, it's a test weirdness and a historical wart of the ARM MPU implementation.

The ztest_test_skip() in question arrived in commit eeab568a277a where it explains "MMU threads within the same memory domain have access to each other's stacks." Which is unambiguously true about mem_domains as documented: all the threads running in the domain get the same view of memory, ergo they can read each others stacks.  QED.

But I guess the early MPU implementations (realistically probably just ARM -- x86 MMU was the second userspace platform, wasn't it?) had some code to isolate stack memory by default, presumably dating from the days before the mem_domain API finalized.  So these tests would pass (i.e. fail to access cross-thread stack memory) on the early MPU implementation.  **But that behavior was never promised or documented AFAICT**.  And of course now you've produced an MPU driver that honors the documented behavior and not "whatever ARM did", so it fails.

Basically: I think this is a historical wart.  These tests are exercising undocumented behavior from an early platform variant that isn't portable to modern Zephyr.  If there's a bug, it's the converse: "legacy MPUs" disallow access to thread stacks that the documentation requires.

I think the correct fix is to remove the tests entirely.  They aren't testing anything but a failure condition.  I'll submit a patch.

My only remaining fear is the dependence on CONFIG_ARCH_MEM_DOMAIN_SYNCHRONOUS_API.  As far as I can tell, that kconfig has absolutely nothing to do with thread stacks or mem_domain.  I **think** (?) you added it as a proxy for "MPU that doesn't do the weird stuff ARM did", am I right?  The actual synchronous API is a driver/platform-level thing.

dcpleung, 2024-03-19:
I guess the question now is... which behavior we want from now on: whether we need to isolate thread stacks from each others in the same memory domain.

andyross, 2024-03-19:
I think we want the MMU behavior for sure, as that matches the way threads in a Unix process work.  Code to share stack pointers across threads isn't that uncommon (c.f. C++ RAII designs might create a manager object on the main thread stack at app startup and then use/update it from a bunch of workers).   Though I guess the Zephyr-side wart/counterargument is that intel_adsp/CONFIG_COHERENCE hands you incoherent stacks where that kind of design becomes a footgun.

One question (and I don't know the answer) is if there's a good hardware reason for ARM being the way it is?  I don't know the ARM MPU code at all, but is there maybe a "free MPU register" you can get to map a stack that you can't use for arbitrary domain memory?  If so we'd be losing something by forcing ARM to use domains as documented.

ceolin, 2024-03-19:
> I think we want the MMU behavior for sure, as that matches the way threads in a Unix process work. Code to share stack pointers across threads isn't that uncommon (c.f. C++ RAII designs might create [...]

Agree, I also prefer MMU behavior. Isolating stacks in the same domain would be very expensive and from security perspective, if an application needs this isolation it can put threads in different domains.

But for me, the fact that we don't enforce this isolation does not mean that we encourage threads accessing other thread stacks, this is in "general" a bad idea. Obviously, this will be a problem with stacks in incoherent memory.

> One question (and I don't know the answer) is if there's a good hardware reason for ARM being the way it is? I don't know the ARM MPU code at all, but is there maybe a "free MPU register" you can [...]

andyross, 2024-03-19:
@carlocaione @stephanosio might have input as to the ARM MPU behavior that informed this design, alas Andrew Boie and Andy Gross aren't with the project anymore.

dcpleung, 2024-04-02:
@carlocaione @stephanosio It seems that the behavior of isolated thread stacks even within the same memory domain predates the introduction of memory domains, hence the different behavior. If we move ARM to the memory domain model where thread stacks can be accessed by every threads within one memory domain, context switching latency can possibly be reduced, since there is no need to re-program the MPU hardware if switching to another thread with same domain. I wonder how much work it will be to convert to the "share-everything within one domain" model.

ruuddw, 2024-04-10:
Hi all, just noticing this thread now. Arm is not the only architecture with MPU, MPUs are very common in MCUs as well as other places where e.g. real-time determinism is important. ARC uses MPU for userspace isolation,
In general, my recommendation would be to have isolated stacks by default, unless there is an explicit reason for sharing. Pointing to another task's stack (which dynamically changes) imho is asking for trouble.
Another argument to stick to the stack isolation is related to implementation efficiency. That depends on how the stacks for the shared memory domain are allocated. Are they in a contiguous memory domain region, or do they need separate MPU entries (scarce resource).
And, there are more complicating factors: are overlapping regions allowed or not? Are stack bound checks enforced by the MPU?
Changing the isolation rules for MPU-based systems is not a simple change.

How about stating task stacks are assumed to be isolated, though if implementation enforcement is expensive then the isolation enforcement may be relaxed in case they all are in the same memory domain?

ceolin, 2024-04-29:
@ruuddw just to be clear, what we are talking here is just about threads in the same memory domain. My understanding is that memory domain is Zephyr's *process* concept. With MMU we would need have a set of page table per thread or change pages attributes when switching threads which is very expensive. I am not familiar with MPU implementations but it seems that it is costly too. If we stick with the current behavior we have two constraints:

1) We don't guarantee stack isolation for threads in the same domain
2) User threads should not access others thread stacks for threads in the same memory domain

ceolin, 2024-05-01:
> @ceolin Sure, I understand the discussion is about threads within a memory domain. Zephyr memory domains are not the same as processes (else, they would have been called processes...). Stacks of userspace threads have been isolated from the start (note MMU-based separation was added after the MPU protection scheme). See the [memory protection](https://docs.zephyrproject.org/latest/kernel/usermode/memory_domain.html) documentation.

I know that memory domains are not the same as process, maybe that was a bad analogy. My point here is that it is expensive for MMUs provide stack isolation for threads in the same memory domain and at same time it seems that is costly to have access to threads in the same domain when using MPU

> 
> > Any thread running in user mode will need access to its own stack buffer. On context switch into a user mode thread, a dedicated MPU region will be programmed with the bounds of the stack buffer.
> 
> and
> 
> > The kernel ensures that any user thread will have access to its own stack buffer, plus program text and read-only data. The memory domain APIs are the way to grant access to additional blocks of memory to a user thread.
 
But if you look the threat model:

```
    - A user thread may have read/write access to the stacks of other user
      threads in the same memory domain, depending on hardware.

       - On MPU systems, threads may only access their own stack buffer.

       - On MMU systems, threads may access any user thread stack in the same
         memory domain. Portable code should not assume this.
```

So if we can't have an unified behavior, I think we shouldn't make any assumption whether or not thread stack isolation is provided for threads in the same memory domain.

The same way one should not assume access to thread stacks in the same domain, it should not assume that they are isolated and this should be consistent across the documentation.

andyross, 2024-05-01:
FWIW, one reason to hate the MPU scheme is that it makes "stacks" a special kind of memory separate from "regular" memory, and we don't track that well.   Like, if you put a r/w partition into your mem_domain that covers your thread stack, does the access work or fail?  It's just a wart.

And to repeat from above: we'll 100% for sure see code ported to Zephyr that uses this paradigm (which works everywhere else in the world) and will eventually need to support both modes on MPU devices anyway.

andyross, 2024-05-01:
And also to clarify @ceolin's point: the fact that this works only on one set of platforms is a security bug by itself.  Having this promise means that we open the possibility of someone porting a "secure" app from ARC or Cortex M to Xtensa or Cortex A or whatever and opening up a hole that they didn't expect.  This is **worse**, not better, than not having made the promise to begin with.

Basically we don't want and shouldn't support both modes.  If we really demand MPU-style thread stack separation then we need to have each thread on a MMU system get its own page tables too to provide the same behavior.

ruuddw, 2024-05-02:
It looks like all of this was already discussed 3 years ago (including discussion and approvals...) #25106, and at that time it seemed the best solution to allow for the different behavior. Did anything change since then?
What was missed is updating the rest of the documentation to be consistent.

andyross, 2024-05-02:
> Did anything change since then?

We realized it's a security bug. :)

aescolar, 2024-05-14:
@dcpleung how is the situation with this issue? Is there a way forward?

dcpleung, 2024-05-14:
> @dcpleung how is the situation with this issue? Is there a way forward?

https://github.com/zephyrproject-rtos/zephyr/pull/71969 provides some kconfigs to indicate whether thread stack isolation with a memory domain is used. That also allows toggling the support if the arch code supports both modes. Currently, choosing one way or the other would require some invasive rewriting. I am not entirely comfortable doing that since next release is LTS.

closed 2024-05-22 by nashif