zephyrproject-rtos/zephyr #70466: efr32xg24_dk2601b  + CONFIG_PM does not compile

marc-hb, 2024-03-20:
**Describe the bug**

As of commit 8f4ac0d4ab45, `west build samples/hello_world/ -p -b efr32xg24_dk2601b -- -DCONFIG_PM=y` does not compile:

> modules/hal/silabs/gecko/service/power_manager/src/sl_power_manager.c:203: undefined reference to `sl_sleeptimer_init'

`CONFIG_PM` enables `modules/hal/silabs/gecko/___/sl_power_manager.c` which invokes `sl_sleeptimer_*` functions from  different .c file(s) in the same git repo.  However `CONFIG_COUNTER_GECKO_STIMER` depends on `DT_HAS_SILABS_GECKO_STIMER_ENABLED` which can only come from... outer space?

**To Reproduce**

`west build samples/hello_world/ -p -b efr32xg24_dk2601b -- -DCONFIG_PM=y`

**Expected behavior**

Compiles.

**Impact**

Breaks new test I'm trying to add:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/8350830496/job/22858189321?pr=70403

Workaround is to exclude efr32xg24_dk2601b

**Environment (please complete the following information):**
Linux, Zephyr SDK, Zephyr commit 8f4ac0d4ab45

**Additional context**

Found in unrelated:
- #70403

marc-hb, 2024-03-20:
Another `silabs` / CONFIG_ failure in the same area:

https://github.com/zephyrproject-rtos/zephyr/actions/runs/8351890993/job/22862465677?pr=70403

```
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c: In function 'sleeptimer_hal_init_timer':
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:73:3: error: unknown type name 'sl_hal_sysrtc_config_t'; did you mean 'sl_sysrtc_config_t'?
   73 |   sl_hal_sysrtc_config_t sysrtc_config = SYSRTC_CONFIG_DEFAULT;
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   sl_sysrtc_config_t
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:3: error: unknown type name 'sl_hal_sysrtc_group_config_t'; did you mean 'sl_sysrtc_group_config_t'?
   74 |   sl_hal_sysrtc_group_config_t group_config = SYSRTC_GROUP_CONFIG_DEFAULT;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   sl_sysrtc_group_config_t
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: error: excess elements in scalar initializer [-Werror]
   74 |   sl_hal_sysrtc_group_config_t group_config = SYSRTC_GROUP_CONFIG_DEFAULT;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: note: (near initialization for 'group_config')
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: error: excess elements in scalar initializer [-Werror]
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: note: (near initialization for 'group_config')
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: error: excess elements in scalar initializer [-Werror]
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: note: (near initialization for 'group_config')
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: error: excess elements in scalar initializer [-Werror]
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: note: (near initialization for 'group_config')
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: error: excess elements in scalar initializer [-Werror]
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:74:47: note: (near initialization for 'group_config')
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:82:3: error: implicit declaration of function 'sl_hal_sysrtc_init'; did you mean 'sl_sysrtc_init'? [-Werror=implicit-function-declaration]
   82 |   sl_hal_sysrtc_init(&sysrtc_config);
      |   ^~~~~~~~~~~~~~~~~~
      |   sl_sysrtc_init
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:84:15: error: request for member 'compare_channel0_enable' in something not a structure or union
   84 |   group_config.compare_channel0_enable = false;
      |               ^
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:86:3: error: implicit declaration of function 'sl_hal_sysrtc_init_group'; did you mean 'sl_sysrtc_init_group'? [-Werror=implicit-function-declaration]
   86 |   sl_hal_sysrtc_init_group(0u, &group_config);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
      |   sl_sysrtc_init_group
/__w/zephyr/modules/hal/silabs/gecko/service/sleeptimer/src/sl_sleeptimer_hal_sysrtc.c:88:3: error: implicit declaration of function 'sl_hal_sysrtc_disable_group_interrupts'; did you mean 'sl_sysrtc_disable_group_interrupts'? [-Werror=implicit-function-declaration]
   88 |   sl_hal_sysrtc_disable_group_interrupts(0u, _SYSRTC_GRP0_IEN_MASK);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   sl_sysrtc_disable_group_interrupts

```

fkokosinski, 2024-03-20:
Hi, thanks for reporting the issue!

> depends on DT_HAS_SILABS_GECKO_STIMER_ENABLED which can only come from... outer space?

`DT_HAS_*_ENABLED` are generated automatically when the corresponding devicetree node is enabled. In this case, [this node](https://github.com/zephyrproject-rtos/zephyr/blob/f0750df3da292bb6d46f05b1a4bb24e5ffc7e0bf/dts/arm/silabs/efr32mg24.dtsi#L136-L143) needs to be enabled for this config to be generated.

> Workaround is to exclude efr32xg24_dk2601b

I'd suggest you exclude the offending boards from the test while this issue is being worked on, as to not stall your PR.

Btw. if your test is doing something PM-related, maybe it would be a good idea to add the `pm` tag to the `testcase.yaml` file?

marc-hb, 2024-03-20:
> In this case, [this node](https://github.com/zephyrproject-rtos/zephyr/blob/f0750df3da292bb6d46f05b1a4bb24e5ffc7e0bf/dts/arm/silabs/efr32mg24.dtsi#L136-L143) needs to be enabled for this config to be generated.

Thanks! When and how does this typically happen?

Would there be a better way to express this dependency than this relatively obscure compilation error? Something in Kconfig maybe?

> Btw. if your test is doing something PM-related, maybe it would be a good idea to add the pm tag to the testcase.yaml file?

What would be the effect? 

The test is not trying to "do" anything related, it's only trying to compile as much .h code as possible.

> I'd suggest you exclude the offending boards from the test while this issue is being worked on, as to not stall your PR.

Already done in #70403, please take a look.

fkokosinski, 2024-04-16:
I've unassigned myself since I'm no longer maintaining SiLabs platorms (see https://github.com/zephyrproject-rtos/zephyr/pull/71219).

aescolar, 2024-04-23:
@fkokosinski unfortunately we need an assignee, as otherwise this issue will keep poping up in bug triage meetings.
Can you recommend anybody, or could you remains as a placeholder?

marc-hb, 2024-06-12:
These compilation failures were fixed when Zephyr commit f4c067c6bcf0fa6 upgraded `hal_silabs` from commit 442d0fb1d02cc4b2bb159fbff378029998b89049 to commit 0c39ee28be31c59a98ed490c3811f68caa1fcbd3.

I have no idea why.

Note this happened _before_ the big HWMv2 rename.

https://github.com/zephyrproject-rtos/hal_silabs/compare/442d0fb1d02cc4b2bb159fbff378029998b89049..0c39ee28be31c59a98ed490c3811f68caa1fcbd3

jhedberg, 2024-06-13:
@marc-hb so we can close this then?

marc-hb, 2024-06-13:
Yes I think we can. Not sure about the process sorry.

jhedberg, 2024-06-13:
Closing, since this is not reproducible anymore