zephyrproject-rtos/zephyr #70478: task_wdt: Watchdog is fed before it is started

lorc-dev, 2024-03-20:
**Describe the bug**
When using the task watchdog subsystem and enabling assertions, the following assert fails:

`ASSERTION FAIL [m_cb[p_instance->drv_inst_idx].state == NRFX_DRV_STATE_POWERED_ON] @ WEST_TOPDIR/modules/hal/nordic/nrfx/drivers/src/nrfx_wdt.c:260`

This looks to be caused by that the watchdog is fed before it is started in [task_wdt.c](https://github.com/zephyrproject-rtos/zephyr/blob/db237875376e3ef548f3691994a7fbb35caef254/subsys/task_wdt/task_wdt.c#L150). The watchdog is started when the first task watchdog is added with `task_wdt_add()`, but `wdt_feed()` is already called in `task_wdt_init()` through `schedule_next_timeout()`.

Looks like the problem was introduced in the following pull request: [#64312](https://github.com/zephyrproject-rtos/zephyr/pull/64312)

This solves the issue:
```diff
diff --git a/subsys/task_wdt/task_wdt.c b/subsys/task_wdt/task_wdt.c
--- a/subsys/task_wdt/task_wdt.c	(revision 468eb56cf242eedba62006ee758700ee6148763f)
+++ b/subsys/task_wdt/task_wdt.c	(date 1710777261929)
@@ -141,6 +141,9 @@
    LOG_ERR("hw_wdt install timeout failed: %d", hw_wdt_channel);
    return hw_wdt_channel;
}
+
+    wdt_setup(hw_wdt_dev, WDT_OPT_PAUSE_HALTED_BY_DBG);
+    hw_wdt_started = true;
#else
return -ENOTSUP;
#endif
```

**Environment:**
 - OS: Windows 10
 - Zephyr v3.6.0
 - Board: nrf52dk_nrf52832

lorc-dev, 2024-05-30:
> I'm not sure if just reverting the previous commit is the right solution, as it would bring back the different issue which was originally solved by that PR.

Yeah, but I doubt the previous issue was really an issue in the task watchdog. 
As the hardware watchdog is NOT yet started in the init function, or at least it shouldn't be. 

My best guess is that it was caused by that the watchdog was started in the `wdt_install_timeout()` function in the stm32 driver, but this is solved by: https://github.com/zephyrproject-rtos/zephyr/pull/53690.
Maybe @arbrauns can confirm that he was using a stm32 controller?

martinjaeger, 2024-06-04:
@lorc-dev I think I prefer the fix in https://github.com/zephyrproject-rtos/zephyr/pull/73699. Can you please check if it works for you as well?

lorc-dev, 2024-06-04:
> @lorc-dev I think I prefer the fix in #73699. Can you please check if it works for you as well?

That works for me.

PR #73699 (2024-06-06): subsys: task_wdt: Feed hardware watchdog only when its started
Previously the schedule_next_timeout() function was feeding the hardware watchdog irrespective of whether or not it was started. This is now fixed.

Fixes #70478
closed 2024-06-06 by carlescufi