zephyrproject-rtos/zephyr #70494: intel_adsp: tests/kernel/semaphore/semaphore/kernel.semaphore fails

majunkier, 2024-03-20:
**Describe the bug**
tested on intel_adsp/ boards (cavs25, ace15_mtpm)

```
tests/kernel/semaphore/semaphore/kernel.semaphore
```

test hangs despite initially promising results. Result is timeout.

**To Reproduce**
` twister --device-testing --hardware-map PATH-TO-HWMAP -s tests/kernel/semaphore/semaphore/kernel.semaphore`

**Expected behavior**
Test should pass

**Logs and console output**
handler.log

```
Running TESTSUITE semaphore
===================================================================
START - test_k_sem_correct_count_limit
PASS - test_k_sem_correct_count_limit in 0.001 seconds
===================================================================
START - test_k_sem_define
PASS - test_k_sem_define in 0.001 seconds
===================================================================
START - test_k_sem_init
PASS - test_k_sem_init in 0.001 seconds
===================================================================
START - test_sem_count_get
PASS - test_sem_count_get in 0.001 seconds
===================================================================
START - test_sem_give_from_isr
PASS - test_sem_give_from_isr in 0.001 seconds
===================================================================
START - test_sem_give_from_thread
PASS - test_sem_give_from_thread in 0.001 seconds
===================================================================
START - test_sem_give_take_from_isr
PASS - test_sem_give_take_from_isr in 0.001 seconds
===================================================================
START - test_sem_measure_timeout_from_thread
PASS - test_sem_measure_timeout_from_thread in 0.001 seconds
===================================================================
START - test_sem_measure_timeouts
PASS - test_sem_measure_timeouts in 1.001 seconds
===================================================================
START - test_sem_multi_take_timeout_diff_sem
SKIP - test_sem_multi_take_timeout_diff_sem in 0.001 seconds
===================================================================
START - test_sem_multiple_threads_wait
PASS - test_sem_multiple_threads_wait in 2.001 seconds
===================================================================
START - test_sem_reset
PASS - test_sem_reset in 0.101 seconds
===================================================================
START - test_sem_reset_waiting
PASS - test_sem_reset_waiting in 0.002 seconds
===================================================================
START - test_sem_take_multiple
PASS - test_sem_take_multiple in 0.941 seconds
===================================================================
START - test_sem_take_no_wait
PASS - test_sem_take_no_wait in 0.001 seconds
===================================================================
START - test_sem_take_no_wait_fails
PASS - test_sem_take_no_wait_fails in 0.001 seconds
===================================================================
START - test_sem_take_timeout
PASS - test_sem_take_timeout in 0.001 seconds
===================================================================
START - test_sem_take_timeout_fails
PASS - test_sem_take_timeout_fails in 0.501 seconds
===================================================================
START - test_sem_take_timeout_forever
PASS - test_sem_take_timeout_forever in 0.101 seconds
===================================================================
START - test_sem_thread2isr
PASS - test_sem_thread2isr in 0.001 seconds
===================================================================
START - test_sem_thread2thread
PASS - test_sem_thread2thread in 0.001 seconds
===================================================================
TESTSUITE semaphore succeeded
Running TESTSUITE semaphore_1cpu
===================================================================
START - test_sem_multiple_take_and_timeouts
SKIP - test_sem_multiple_take_and_timeouts in 0.002 seconds
===================================================================
START - test_sem_queue_mutual_exclusion

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version used: v3.6.0-1236-g8f4ac0d4ab45

nashif, 2024-03-20:
@kv2019i  FYI

peter-mitsis, 2024-03-20:
For what it is worth, I have a theory that may explain the reported failure.

**Background**
The failing test (test_sem_queue_mutual_exclusion) is part of the _semaphore_1cpu_ test suite. This test suite "disables" other CPUs. It does this by having those other CPUs busy wait with interrupts locked for a long period of time.

**The Theory**
I suspect that core 0, the only core to which timer interrupts are being delivered, is one of those CPUs that has interrupts locked for a long duration. If this is the case, then the timer interrupt is never processed by core 0.

andyross, 2024-03-20:
I was a little worried about this in the original PR.  There's a lot of weird assumptions in some of the first era of tests that were ported to SMP, and broadcast timer interrupts were a requirement for a long time.  (Even now it's a little ambiguous, riscv64 for example will deliver the timer interrupt on the CPU where it happened to be scheduled, leaving any other scheduled interrupts still pending on other CPUs, which is fine too, but neither is it this "cpu0 only" scheme or broadcast).

FWIW: given the one-line nature of the patch, seems like a easy fix would be to define a CONFIG_SYS_CLOCK_MP_BROADCAST (or whatever) kconfig that selects whether or not the timer is broadcast, just gate that line with an #if, and set it for whatever test cases are failing.   Extra credit would put it into the platform layers as a "HAS_BROADCAST" flag they could expose maybe, etc...

PR #70579 (2024-03-26): tests: z_test_1cpu_start() makes only CPU0 active
When z_test_1cpu_start() is called to ensure that only a single CPU on an SMP system is available for use in a test, this commit will ensure that that CPU is the primary CPU--CPU0. This is done because some timer drivers only have the timer interrupt processed by one CPU.

A bit of a song and dance is performed to achieve this without enabling the CPU mask/affinity pinning API. If the cpuhold thread is found to be executing on CPU0, then a new copy of cpuhold thread is created. Once the new copy is executing (incidentally guaranteed to be on an [... cut]
closed 2024-03-26 by dleach02