zephyrproject-rtos/zephyr #70496: MPU: execute code from RAM

Maartenwn, 2024-03-20:
**Is your enhancement proposal related to a problem? Please describe.**
Currently there seems to be no way to execute code from RAM with the MPU enabled. 

Some context: Our product is a game console. We implemented code that loads the games from an filesystem into RAM. We predefined a region so the code is always executed from the same location. During the development we disabled the MPU however we would like to use the MPU for security reasons. There currently is no memory-region attribute with both read, write and executable access without enabling this for all flash (by enabling the kconfig symbol CONFIG_MPU_ALLOW_FLASH_WRITE).

**Describe the solution you'd like**
We found a solution by adding an extra memory-region helper in include/zephyr/arch/arm/mpu/arm_mpu_v7m.h with read, write and execute access. This allowed us to create a custom memory-region in the device tree, see below.
```
GAMES_FLASH: memory@20038000 {
        compatible = "zephyr,memory-region";
        reg = <0x20038000 0x8000>;
        zephyr,memory-region = "GAMES_FLASH";
        zephyr,memory-region-mpu = "RAM_EXECUTABLE";
    };
``` 
Noted we are a few zephyr versions behind. 

We wonder if this solution is generic enough to be included in Zephyr?
Or would a more elaborate solution be preferred?

**Describe alternatives you've considered**
We have tried using USERSPACE however this added too much overhead (RAM and FLASH usage) for our solution.

TerryGeng, 2025-03-23:
Similar issues were reported for RP2040 as well: https://github.com/zephyrproject-rtos/zephyr/issues/54545. The workaround is to allow code execution from SRAM when configuring the MPU (by removing the `MPU_RASR_XN_Msk` flag): https://github.com/zephyrproject-rtos/zephyr/blob/961593d1529002b33cb568499090c111584d7c42/include/zephyr/arch/arm/mpu/arm_mpu_v7m.h#L120-L122.

In the pico-sdk, functions in the SRAM are stored in the `.time_critical` section. I wonder if there is a way to create a new MPU region from the address of this section?

BujSet, 2025-07-17:
Do you have a work around for the `mp3_corstone300_fvp` board? It uses `arm_mpu_v8.h` header. I'm guessing it might involve changes to [here](https://github.com/zephyrproject-rtos/zephyr/blob/961593d1529002b33cb568499090c111584d7c42/include/zephyr/arch/arm/mpu/arm_mpu_v8.h#L246)?