zephyrproject-rtos/zephyr #70508: Build with ARC MWDT became broken after HMWv2 was implemented.

kokas-a, 2024-03-20:
**Describe the bug**
Build with ARC MWDT became broken after HMWv2 was implemented.

ARC MWDT toolchain doesn't contain it's own DTC-preprocessor. It uses preprocessor, which is a part of Zephyr-SDK.
In current implementailon ```${ZEPHYR_SDK_INSTALL_DIR}/cmake/zephyr/target.cmake``` file should be included on very early 
build stages to obtain the preprocessor via CROSS_COMPILE variable (it defined inside pointed .cmake file).
Modules call sequence: 
```zephyr_default->dtc->FindHostTools->toolchain/arcmwdt/generic.cmake->${ZEPHYR_SDK_INSTALL_DIR}/cmake/zephyr/target.cmake```

The point is target.cmake requires ARCH variable to be defined. New HWMv2 defines ARCH variable much more later in arch_v2 module via CONFIG_ARCH.

In case of MWDT such construction as ```set(ARCH ${CONFIG_ARCH})``` is not applicable as Kconifg module was not yes initialized. And it at the same time requires dtc, and we got circullar dependence. 

So, here I'd like to ask for a advice how to fix the issue in the correct way.

**To Reproduce**
```Steps to reproduce the behavior:
1. export ZEPHYR_TOOLCHAIN_VARIANT=arcmwdt
2. west build -b nsim/nsim_em samples/hello_world/
3. 
-- Board: nsim, identifier: nsim_em
-- ARCMWDT_TOOLCHAIN_PATH is not set, use default toolchain from METAWARE_ROOT: '/home/user/ARC_202312/MetaWare'
-- Found host-tools: zephyr 0.16.3 (/home/user/sandbox/zephyr-sdk-0.16.3)
-- Found dts preprocessor: CMAKE_DTS_PREPROCESSOR-NOTFOUND (Zephyr SDK 0.16.3)
```

**Expected behavior**
Build not fails:
```
...
-- Found host-tools: zephyr 0.16.3 (/home/user/sandbox/zephyr-sdk-0.16.3)

CMake Debug Log at /home/user/sandbox/zephyr-sdk-0.16.3/cmake/zephyr/target.cmake:32 (if):
  Variable "ARCH" was accessed using READ_ACCESS with value "arc".
Call Stack (most recent call first):
  /home/user/sandbox/zephyrproject/zephyr/cmake/toolchain/arcmwdt/generic.cmake:26 (include)
  /home/user/sandbox/zephyrproject/zephyr/cmake/modules/FindHostTools.cmake:103 (include)
  /home/user/sandbox/zephyrproject/zephyr/cmake/modules/dts.cmake:9 (find_package)
  /home/user/sandbox/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /home/user/sandbox/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/user/sandbox/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Found dts preprocessor: /home/user/sandbox/zephyr-sdk-0.16.3/arc-zephyr-elf/bin/arc-zephyr-elf-gcc (Zephyr SDK 0.16.3)
...
```

**Impact**
Showstopper

**Logs and console output**
```Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/sandbox/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.25.1
-- Found Python3: /home/user/sandbox/zephyrproject/.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/user/sandbox/zephyrproject/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: nsim, identifier: nsim_em
-- ARCMWDT_TOOLCHAIN_PATH is not set, use default toolchain from METAWARE_ROOT: '/home/user/ARC_202312/MetaWare'
-- Found host-tools: zephyr 0.16.3 (/home/user/sandbox/zephyr-sdk-0.16.3)
CMake Debug Log at /home/user/sandbox/zephyr-sdk-0.16.3/cmake/zephyr/target.cmake:32 (if):
  Variable "ARCH" was accessed using UNKNOWN_READ_ACCESS with value "".
Call Stack (most recent call first):
  /home/user/sandbox/zephyrproject/zephyr/cmake/toolchain/arcmwdt/generic.cmake:26 (include)
  /home/user/sandbox/zephyrproject/zephyr/cmake/modules/FindHostTools.cmake:104 (include)
  /home/user/sandbox/zephyrproject/zephyr/cmake/modules/dts.cmake:9 (find_package)
  /home/user/sandbox/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /home/user/sandbox/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/user/sandbox/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Found dts preprocessor: CMAKE_DTS_PREPROCESSOR-NOTFOUND (Zephyr SDK 0.16.3)
-- Found Dtc: /home/user/sandbox/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/user/sandbox/zephyrproject/zephyr/boards/synopsys/nsim/nsim_nsim_em.dts
CMake Error at /home/user/sandbox/zephyrproject/zephyr/cmake/modules/extensions.cmake:4205 (message):
  failed to preprocess devicetree files (error code No such file or
  directory)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: arcmwdt
 - Commit SHA or Version used: 1a55caf8263e11bc22f93b88e2ccf99eb30f690f

**Additional context**

aescolar, 2024-04-30:
@tejlmand or @nordicjm , @57300  can you please take a look at this?

aescolar, 2024-05-14:
@kokas-a After https://github.com/zephyrproject-rtos/zephyr/pull/71163 was merged, can this be closed?

kokas-a, 2024-05-14:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/71163