zephyrproject-rtos/zephyr #70509: Version: add git describe flag --dirty to mark if build was with come thanges on build commit

Aukstkalnis, 2024-03-20:
I would like to use default zephyr's version generation functionality in my application but it is missing one little feature that can help to troubleshoot hidden problems such as code that had some extra changes on commit that was used to build application. 

**Describe the solution you'd like**
When zephyr's tool chain generates app_version (or zephyr's version ) it can use --dirty git describe command's flag to mark the build as having uncommitted changes when building. This flag can add -dirty on commits hash or it can add specific text or symbol such as '+' (--dirty=+ and the result will be <hash>+). It can be configurable functionality.

**Describe alternatives you've considered**
My alternative was to just copy version generation cmake functionality to my application root directory and create my own version of generating the same version.h file but with little + sign at the end of the has.

**Additional context**
So my version looks like this when commit has uncommitted changes:
#define BUILD_APP_VERSION "6dfcc1600a19+"
And looks like this when no uncommitted changes are present:
#define BUILD_APP_VERSION "6dfcc1600a19"

Interestingly, BUILD_VERSION in zephyr is defined without the quotes so I modified the  app_version.h generation to add them. It is easier to use this define as a string then number with text that you need to call special macro to make a string out of this define. Because it is not a number (there could be symbols from a-f) and not  a string (no quotes) :)

Aukstkalnis, 2024-11-13:
Yes, you are right. But what if you will change .c file in application's directory? I think this will be visible with with --dirty flag. Am I right? But today we will not know if anything is dirty at all. So I think this still will be better then it is now.

And I think that zephyr and all the other modules are git repositories. And all can be tested with --dirty flag. And you can document all the limitations.

nordicjm, 2024-11-13:
> Yes, you are right. But what if you will change .c file in application's directory? I think this will be visible with --dirty flag. Am I right? But today we will not know if anything is dirty at all. So I think this still will be better then it is now.

It will not, no, because cmake is not tracking your .c file, git is, so your application output will be updated but the generated app_version.h file will not be, hence hitting the issue I described above. You can even try it out now, build your project with no changes, modify a .c file, then check the modification time of the app_version.h file, notice how it has not changed

bearsh, 2024-11-18:
I've solved this by adding a custom target (which is a dependency of the application) to cmake which re-generates the VERSION file if needed. The script doing this could also check all west deps by calling e.g. `west status`

nordicjm, 2024-11-18:
> I've solved this by adding a custom target (which is a dependency of the application) to cmake which re-generates the VERSION file if needed. The script doing this could also check all west deps by calling e.g. `west status`

A very quick way to completely bypass the whole check, assuming you have this in your cmake file:
```
file(GLOB app_sources src/*.c)
target_sources(app PRIVATE ${app_sources})
```
Add a new file e.g. `src/somefilehere.c` with:
```
static int new_function(void)
{
        printk();

        return 0;
}
SYS_INIT(new_function, APPLICATION, 0);
```
The application output is completely different from what's in the git repo, yet it will show as being completely clean.