zephyrproject-rtos/zephyr #70516: Ci regression from  #69746

fabiobaltieri, 2024-03-20:
**Describe the bug**
https://github.com/zephyrproject-rtos/zephyr/pull/69746 broke the nanopb build in CI for most platforms:

```
$ west build -p -b nrf52dk/nrf52832 samples/modules/nanopb
...
CMake Error at /usr/local/google/home/fabiobaltieri/zephyrproject/modules/lib/nanopb/extra/FindNanopb.cmake:407 (target_compile_features):
  target_compile_features The compiler feature "c_std_11" is not known to C
  compiler

  "GNU"
```

https://github.com/zephyrproject-rtos/zephyr/actions/runs/8364553686/job/22900251867

**To Reproduce**
`west build -p -b nrf52dk/nrf52832 samples/modules/nanopb` on main

**Expected behavior**
Not break.

**Impact**
The "push" CI run is very red, does not seem to be impacting pull request targets.

fabiobaltieri, 2024-03-20:
@nashif @aescolar @tejlmand  fyi, unsure whether this should be rolled back or if a fix forward that does not involve the module can be produced, looking at `cmake/compiler/compiler_features.cmake` it feels like this should work somewhat but I can't really tell what's wrong.

pdgendt, 2024-03-20:
@fabiobaltieri How come this didn't fail the CI in the PR itself?

pdgendt, 2024-03-20:
It didn't run any CI?

henrikbrixandersen, 2024-03-20:
I would roll back the manifest change.

closed 2024-03-20 by fabiobaltieri

fabiobaltieri, 2024-03-20:
> It didn't run any CI?

Yeah I guess so, the pull request and push runs have a different set of tests, not super familiar with the details but it's a known shortcoming and there's some effort to rework it. Unfortunately in the current situation this stuff happens.

PR #70518 (2024-03-20): Revert "manifest: optional: Update nanopb"
Upstream repo has made changes to the cmake module, which breaks `native_sim` builds.

This reverts commit d737e5d8e3606b1e1879e03535cf7dcf34b83317.

Will look into a proper fix.

Fixes #70516