zephyrproject-rtos/zephyr #70535: audio framework

hakehuang, 2024-03-21:
**Is your feature request related to a problem? Please describe.**

in zephyr, we do not have good framework to organize the audio process such as:
1. recording
2. playback
3. audio mixing
4. audio recognization
5. audio noise reduction
6. audio codec output
7. audio encode and decoder
8. stream playback thought ethernet/ble
9. audio firware deploy and upgrading
10. low power for audio
11. a good test approach to audio

now we have to in application layer to assembly all of them, by using the i2s driver.  see this PR
https://github.com/zephyrproject-rtos/zephyr/pull/66609 

**Describe the solution you'd like**
any plan to create a unified support for audio stuff in Zephyr?

**Describe alternatives you've considered**
ALSA/OpenSound in Linux, but it is portable to Zephyr RTOS?

hakehuang, 2024-03-21:
@golowanow any commnets?

hongshui3000, 2024-03-21:
This is another similar requirement that I mentioned before
https://github.com/zephyrproject-rtos/zephyr/issues/58098

closed 2024-03-25 by hakehuang