zephyrproject-rtos/zephyr #70539: Release of touch input device results in wrong coordinates while using "invert-x/y" (lvgl-pointer-input) due to double inversion of coordinates

faxe1008, 2024-03-21:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/70500

<div type='discussions-op-text'>

<sup>Originally posted by **Benni77** March 20, 2024</sup>
I'm wondering if this is a bug with request for confirmation before I open an issue:

I'm using a gc9a01 display with [CST816S](https://docs.zephyrproject.org/latest/build/dts/api/bindings/input/hynitron%2Ccst816s.html) touch controller with lvgl.
When I use the lvgl [Dropdown Widget](https://docs.lvgl.io/master/examples.html#dropdown) the selection of the list options is not working correctly.
Digging into it made me realize this only happens if `invert-x` or `invert-y` (propably with `swap-xy` as well) configuration is activated in [zephyr,lvgl-pointer-input ](https://docs.zephyrproject.org/latest/build/dts/api/bindings/input/zephyr%2Clvgl-pointer-input.html) device:
```
lvgl_pointer {
     compatible = "zephyr,lvgl-pointer-input";
     input = <&cst816s>;
     invert-x;
     invert-y;
};
```

So i digged deeper:
In the input drivers [input_cst816s.c](https://github.com/zephyrproject-rtos/zephyr/blob/d737e5d8e3606b1e1879e03535cf7dcf34b83317/drivers/input/input_cst816s.c#L107) processing function (cst816s_process()) a input is reported on release of the touch device in the else statement:
```c
	if (pressed) {
		input_report_abs(dev, INPUT_ABS_X, col, false, K_FOREVER);
		input_report_abs(dev, INPUT_ABS_Y, row, false, K_FOREVER);
		input_report_key(dev, INPUT_BTN_TOUCH, 1, true, K_FOREVER);
	} else {
	        input_report_key(dev, INPUT_BTN_TOUCH, 0, true, K_FOREVER);
	}
```

Now in the `lvgl_pointer_process_event()` function in the [lvgl_pointer_input.c](https://github.com/zephyrproject-rtos/zephyr/blob/d737e5d8e3606b1e1879e03535cf7dcf34b83317/modules/lvgl/input/lvgl_pointer_input.c#L58) file the coordinates of the input event get inverted:
```c
	if (cfg->invert_x) {
		if (cap->current_orientation == DISPLAY_ORIENTATION_NORMAL ||
		    cap->current_orientation == DISPLAY_ORIENTATION_ROTATED_180) {
			point->x = cap->x_resolution - point->x;
		} else {
			point->x = cap->y_resolution - point->x;
		}
	}

	if (cfg->invert_y) {
		if (cap->current_orientation == DISPLAY_ORIENTATION_NORMAL ||
		    cap->current_orientation == DISPLAY_ORIENTATION_ROTATED_180) {
			point->y = cap->y_resolution - point->y;
		} else {
			point->y = cap->x_resolution - point->y;
		}
	}
```
**Problem**: Since this inversion also happens on the "RELEASE" event without new coordinates passed by the input device this results in a double inversion of the last coordinates: First when new coordinates are passed, second when "RELEASE" is passed.

**Result**: Since the Dropdown widget uses the double inverted coordinates from the "RELEASE" event, this unfortunately results in a malfunction of the widget. The list items can not be selected correctly.

For me it looks like a bug in the `lvgl_pointer_process_event()` function, which should not invert on "RELEASE" event, what do you guys think?

Cheers
Benjamin</div>

fabiobaltieri, 2024-03-26:
Yeah it's a bug in `no prob that is `, thanks for debugging it.

I'm thinking that the easiest way of handling this would be to refactor the point inversion and rotation into the `switch` statement, so that the data only gets processed when the raw point arrives. @faxe1008 what do you think?

faxe1008, 2024-04-30:
Resolved via #70541

closed 2024-04-30 by faxe1008