zephyrproject-rtos/zephyr #70551: Bluetooth: Audio: Move channel allocation strings to a public header

koffes, 2024-03-21:
In https://github.com/zephyrproject-rtos/zephyr/blob/2396b01b1ea2b0b6a61591155508cf19cd9d367c/subsys/bluetooth/audio/shell/audio.h#L714
There is a nice function, returning strings for all the channel allocations.

We would like to have this moved to a public location, and have a guard in place so only one bit can be set in the location bitfield sent to the function. If more than one bit is set it could return "Invalid" or something similar.

kruithofa, 2024-03-21:
As per Nick Hunns  'Introducing Bluetooth LE Audio', chapter 3.6: 

> A single channel speaker would normally set both the Front Left and Front Right audio locations

So more than one bit can be set. @Thalley do you remember the corresponding spec, and if this is still true?

koffes, 2024-03-21:
Alternatively, and perhaps better, the function could take a pointer to a string buffer and that buffer size. Then it can iterate and populate up to the given size. If it goes over, it could write "..." on the last indexes before the NUL terminator.

Thalley, 2024-03-21:
If we want to support multiple bits, then we need a buffer in RAM, whereas just handling a single bit per call is effectively only in flash. 

Different applications may also log or print them differently (e.g. the shell prints them indented with one value per line). 

I suggest we keep it simple and just returning a string for a specific bit, and then the application will be in a much better condition to use this as they see fit.

Thalley, 2024-05-07:
> I would like to do this ticket. My plan to is to move the function to `subsys/bluetooth/audio/audio.c` and rename it to something like
> 
> ```c
> char *bt_audio_chan_location_bit_to_str(enum bt_audio_location chan_allocation)
> ```
> 
> ?
> 
> Should I add a test case for when someone attempts to set more then one bit?

That would be great :) 

While this issue only requests the channel allocation, there is actually quite a few others that could benefit from the same treatment: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/shell/audio.h#L270-L834

I suggest to add the public API string function for all of these (some of them are easier than location, as they are not bitfields). 

> Should I add a test case for when someone attempts to set more then one bit?

I think that may be tricky to test. What do you think the behavior should be in that case? Ideally the function will return a `const char *`, so on error it could 

1. Return NULL
2. Return "Invalid input" (or similar)
3. Return the first bit set
4. Assert

Thalley, 2024-05-07:
> > I think that may be tricky to test. What do you think the behavior should be in that case? Ideally the function will return a `const char *`, so on error it could
> > 
> > 1. Return NULL
> > 2. Return "Invalid input" (or similar)
> > 3. Return the first bit set
> > 4. Assert
> 
> I like returning NULL or "Invalid Input". IMO returning the first bit set is not the correct behaviour and Asserting would be overkill. I will go with NULL unless you want "Invalid Input". I wont do the test case.

Biggest issue with returning NULL, is that you always have the check the output before printing/logging the return value, so in that case returning an error string may be better. 
> 
> > While this issue only requests the channel allocation, there is actually quite a few others that could benefit from the same treatment: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/shell/audio.h#L270-L834
> 
> Are you okay with me doing this all in one PR?

Absolutely! :)

Thalley, 2024-05-29:
@Ballen7 any progress on this?