zephyrproject-rtos/zephyr #70557: samples/drivers/can/counter/  test not working

marwaiehm-st, 2024-03-21:
**Describe the bug**
The Can counter  sample does not work on the board stm32h573i_dk

**To Reproduce**
Compile the  Can counter sample for the stm32h573i_dk platform using this command:
```shell
west build -p auto -b  stm32h573i_dk samples/drivers/can/counter/
``` 

**Expected behavior**
The sample should show this message : "Counter received: x"
```shell

.*** Booting Zephyr OS build v3.6.0-1359-g50542682df94 ***
Change LED filter ID: 0
Finished init.
Counter filter id: 28

uart:~$ Counter received: 0
Counter received: 1
Counter received: 2
Counter received: 3
Counter received: 4
Counter received: 5
Counter received: 6
Counter received: 7
Counter received: 8
Counter received: 9
Counter received: 10
Counter received: 11
Counter received: 12
Counter received: 13
Counter received: 14
Counter received: 15
Counter received: 16
Counter received: 17
Counter received: 18
Counter received: 19
Counter received: 20
Counter received: 21

``` 

**Impact**
The test is blocked, there is no  reception of the output "received counter: values" due to this commit: https://github.com/zephyrproject-rtos/zephyr/commit/caf4a2933896dedff02b478ca67944077d8aad5a : 
The modification of  the **APB1** bus clock to 120MHZ cause  **FDCAN** bus reception problem,  because they are using the same bus clock and the **FDCAN** bus needs more then 120MHZ to work correctly.

**Logs and console output**
RTT log output:
```shell
*** Booting Zephyr OS build v3.6.0-1359-g50542682df94 ***
Change LED filter ID: 0
Finished init.
Counter filter id: 28

uart:~$ 
``` 

**Environment :**
 - OS:Linux
 - Toolchain  Zephyr SDK v3.6.0

str4t0m, 2024-03-23:
I do think the issue is that APB1 has a frequency of 120MHz specifically, but that issue is that fdcan_clk(in stm32 terms) supplied via PLL1_Q is higher than APB1.
M_CAN requires that the host clock(APB1) is higher or equal to the CAN core clock.
So setting the PLL1_Q divider to 6 giving a can core clock of 80MHz should solve the issue and also result in good interoperability with other nodes.

henrikbrixandersen, 2024-03-23:
I agree with what @str4t0m suggested. Afterwards, you can verify the timing using `tests/drivers/can/timing`.

erwango, 2024-04-02:
@marwaiehm-st Can we close the issue, now that https://github.com/zephyrproject-rtos/zephyr/pull/70569 was merged ?

closed 2024-04-02 by marwaiehm-st