zephyrproject-rtos/zephyr #70582: llext: sem EXPORT_SYMBOL not working in userspace

laurenmurphyx64, 2024-03-21:
**Describe the bug**
Discovered on ARM qemu e.g. mps2/an385 while working on #70317, reproduceable with [this branch](https://github.com/laurenmurphyx64/zephyr/commit/5835b2fd276f7a8e0008032b3069ae86e8f8a8f1) and

```
west build -p -b mps2/an385 tests/subsys/llext/simple -- -DCONFIG_USERSPACE=y  && west build -t run
```

EXPORT_SYMBOL for k_thread_join, etc. working, semaphore not

**Expected behavior**
Export succeeds 

**Impact**
Showstopper 

**Logs and console output**
I printed out the contents of the built-in symbol table with and without userspace.

Without userspace:
```
assert_post_action
assert_print
k_is_in_isr
memcmp
memcpy
memset
printk
strcmp
strcpy
strlen
strncmp
strncpy
vprintk
z_impl_k_sem_give
z_impl_k_sem_init
z_impl_k_sem_take
z_impl_k_thread_create
z_impl_k_thread_join
z_impl_k_thread_start
z_log_minimal_printk
ztest_relative_filename
ztest_test_fail
ztest_test_pass
ztest_test_skip
```

With userspace:
```
assert_post_action
assert_print
k_is_in_isr
memcmp
memcpy
memset
printk
strcmp
strcpy
strlen
strncmp
strncpy
vprintk
z_arm_thread_is_in_user_mode
z_impl_k_thread_create
z_impl_k_thread_join
z_impl_k_thread_start
z_log_minimal_printk
z_mrsh___posix_clock_get_base
z_mrsh___ztest_set_test_phase
z_mrsh___ztest_set_test_result
z_mrsh_device_get_binding
z_mrsh_device_is_ready
z_mrsh_gpio_get_pending_int
z_mrsh_gpio_pin_configure
z_mrsh_gpio_pin_get_config
z_mrsh_gpio_pin_interrupt_configure
z_mrsh_gpio_port_clear_bits_raw
z_mrsh_gpio_port_get_direction
z_mrsh_gpio_port_get_raw
z_mrsh_gpio_port_set_bits_raw
z_mrsh_gpio_port_set_masked_raw
z_mrsh_gpio_port_toggle_bits
z_mrsh_k_busy_wait
z_mrsh_k_condvar_broadcast
z_mrsh_k_condvar_init
z_mrsh_k_condvar_signal
z_mrsh_k_condvar_wait
z_mrsh_k_event_clear
z_mrsh_k_event_init
z_mrsh_k_event_post
z_mrsh_k_event_set_masked
z_mrsh_k_event_set
z_mrsh_k_event_wait_all
z_mrsh_k_event_wait
z_mrsh_k_float_disable
z_mrsh_k_float_enable
z_mrsh_k_futex_wait
z_mrsh_k_futex_wake
z_mrsh_k_is_preempt_thread
z_mrsh_k_msgq_alloc_init
z_mrsh_k_msgq_get_attrs
z_mrsh_k_msgq_get
z_mrsh_k_msgq_num_free_get
z_mrsh_k_msgq_num_used_get
z_mrsh_k_msgq_peek_at
z_mrsh_k_msgq_peek
z_mrsh_k_msgq_purge
z_mrsh_k_msgq_put
z_mrsh_k_mutex_init
z_mrsh_k_mutex_lock
z_mrsh_k_mutex_unlock
z_mrsh_k_object_access_grant
z_mrsh_k_object_alloc_size
z_mrsh_k_object_alloc
z_mrsh_k_object_release
z_mrsh_k_pipe_alloc_init
z_mrsh_k_pipe_buffer_flush
z_mrsh_k_pipe_flush
z_mrsh_k_pipe_get
z_mrsh_k_pipe_put
z_mrsh_k_pipe_read_avail
z_mrsh_k_pipe_write_avail
z_mrsh_k_poll_signal_check
z_mrsh_k_poll_signal_init
z_mrsh_k_poll_signal_raise
z_mrsh_k_poll_signal_reset
z_mrsh_k_poll
z_mrsh_k_queue_alloc_append
z_mrsh_k_queue_alloc_prepend
z_mrsh_k_queue_cancel_wait
z_mrsh_k_queue_get
z_mrsh_k_queue_init
z_mrsh_k_queue_is_empty
z_mrsh_k_queue_peek_head
z_mrsh_k_queue_peek_tail
z_mrsh_k_sched_current_thread_query
z_mrsh_k_sem_count_get
z_mrsh_k_sem_give
z_mrsh_k_sem_init
z_mrsh_k_sem_reset
z_mrsh_k_sem_take
z_mrsh_k_sleep
z_mrsh_k_stack_alloc_init
z_mrsh_k_stack_pop
z_mrsh_k_stack_push
z_mrsh_k_str_out
z_mrsh_k_thread_abort
z_mrsh_k_thread_create
z_mrsh_k_thread_custom_data_get
z_mrsh_k_thread_custom_data_set
z_mrsh_k_thread_deadline_set
z_mrsh_k_thread_join
z_mrsh_k_thread_name_copy
z_mrsh_k_thread_name_set
z_mrsh_k_thread_priority_get
z_mrsh_k_thread_priority_set
z_mrsh_k_thread_resume
z_mrsh_k_thread_stack_alloc
z_mrsh_k_thread_stack_free
z_mrsh_k_thread_stack_space_get
z_mrsh_k_thread_start
z_mrsh_k_thread_suspend
z_mrsh_k_thread_timeout_expires_ticks
z_mrsh_k_thread_timeout_remaining_ticks
z_mrsh_k_timer_expires_ticks
z_mrsh_k_timer_remaining_ticks
z_mrsh_k_timer_start
z_mrsh_k_timer_status_get
z_mrsh_k_timer_status_sync
z_mrsh_k_timer_stop
z_mrsh_k_timer_user_data_get
z_mrsh_k_timer_user_data_set
z_mrsh_k_uptime_ticks
z_mrsh_k_usleep
z_mrsh_k_wakeup
z_mrsh_k_yield
z_mrsh_log_buffered_cnt
z_mrsh_log_filter_set
z_mrsh_log_frontend_filter_set
z_mrsh_log_panic
z_mrsh_log_process
z_mrsh_sys_clock_hw_cycles_per_sec_runtime_get
z_mrsh_sys_clock_tick_set
z_mrsh_uart_config_get
z_mrsh_uart_configure
z_mrsh_uart_drv_cmd
z_mrsh_uart_err_check
z_mrsh_uart_irq_err_disable
z_mrsh_uart_irq_err_enable
z_mrsh_uart_irq_is_pending
z_mrsh_uart_irq_rx_disable
z_mrsh_uart_irq_rx_enable
z_mrsh_uart_irq_tx_disable
z_mrsh_uart_irq_tx_enable
z_mrsh_uart_irq_update
z_mrsh_uart_line_ctrl_get
z_mrsh_uart_line_ctrl_set
z_mrsh_uart_poll_in
z_mrsh_uart_poll_in_u16
z_mrsh_uart_poll_out
z_mrsh_uart_poll_out_u16
z_mrsh_uart_rx_disable
z_mrsh_uart_rx_enable
z_mrsh_uart_rx_enable_u16
z_mrsh_uart_tx_abort
z_mrsh_uart_tx
z_mrsh_uart_tx_u16
z_mrsh_z_log_msg_simple_create_0
z_mrsh_z_log_msg_simple_create_1
z_mrsh_z_log_msg_simple_create_2
z_mrsh_z_log_msg_static_create
z_mrsh_z_sys_mutex_kernel_lock
z_mrsh_z_sys_mutex_kernel_unlock
z_mrsh_z_test_1cpu_start
z_mrsh_z_test_1cpu_stop
z_mrsh_zephyr_fputc
z_mrsh_zephyr_fwrite
z_mrsh_zephyr_read_stdin
z_mrsh_zephyr_write_stdout
z_mrsh_ztest_run_test_suites
z_mrsh_ztest_set_assert_valid
z_mrsh_ztest_set_fault_valid
ztest_relative_filename
ztest_test_fail
ztest_test_pass
ztest_test_skip
```

**Environment (please complete the following information):**
 - OS: Ubuntu 22
 - Toolchain: Zephyr SDK
 - `3b36911`

laurenmurphyx64, 2024-03-21:
Tagging in @edersondisouza @teburd mentioned you recently made an update to export all syscalls in userspace

edersondisouza, 2024-03-21:
Hi @laurenmurphyx64 - can you check if reverting 67bb6db3f8e5329516f005f1335bb7a23a95b3eb does help?

laurenmurphyx64, 2024-03-21:
> Hi @laurenmurphyx64 - can you check if reverting [67bb6db](https://github.com/zephyrproject-rtos/zephyr/commit/67bb6db3f8e5329516f005f1335bb7a23a95b3eb) does help?

Reverted, doesn't help. I noticed this on a branch created before you merged that - probably should've mentioned it.

edersondisouza, 2024-03-21:
Aha, know what is going on. Stay tuned, I'll have a PR that can help - tested here, and it works for your branch.

PR #70584 (2024-03-26): llext export syscalls fixes
With 67bb6db3f8e5329516f005f1335bb7a23a95b3eb, one doesn't need to export manually each syscall available. However, there are some flaws there.

First is that it only works if the extension lives on userspace. It was exporting the `z_mrsh_` symbols, available to userspace. If an extension actually lives in kernelspace, it wouldn't have the `z_impl_` symbol available.

Second flaw is that syscalls could still be optimised away if the Zephyr application itself wasn't using them. This is the issue of #70582 - and it was there before 67bb6db3f8e5329516f005f1335bb7a23a95 [... cut]
closed 2024-03-26 by nashif