zephyrproject-rtos/zephyr #70586: drivers: dma: `block_size` API parameter ambiguity

gramsay0, 2024-03-22:
**Describe the bug**
`struct dma_block_config` specifies `block_size` as: "Number of bytes to be transferred for this block"
https://github.com/zephyrproject-rtos/zephyr/blob/105ef05c52d62fbb192eecb78475567d8475db0f/include/zephyr/drivers/dma.h#L126C6-L126C54

This does not make sense in some situations, since the source and destination data width may not be the same (`source_data_size`/`dest_data_size`).

e.g. if you transfer 100 bytes into 100 32bit words, is that 100 or 400 bytes transferred?

Renaming `block_size`  -> `block_len` and documenting it as the number of transfers (rather than bytes) may make more sense.

**Impact**
Confusion when implementing or using the DMA API

**Additional context**
[Discord chat](https://discord.com/channels/720317445772017664/1038098733130326017/1220496622157565992)

@teburd

teburd, 2024-04-09:
After reviewing some dmac IP docs and implementations we have I'm inclined to disagree with the idea of block_len implying transfers as a term here. The number of transfers is hardware dependent and not necessarily something an API trying to account for many DMA controllers should directly expose as a configuration parameter... it may not actually be configurable by software.

NXP's eDMA transfer descriptor has a field for the transfer size in bytes.
Synopsis's DMAC transfer descriptor likewise has a field for the transfer size in bytes.
Atmels XDMAC is a little more confusing to look at but talks of micro transfers of burst length size with non-integral match ups leading to transfers to account for this.

In all cases the number of actual transfers as I'd view them as bus mastering events is entirely hardware dependent. We don't know how many transfers are done. We simply know that We want to move N bytes from a source address, with address adjustment and address word sizing.

I will say that the block size should probably be a multiple of the source word size... but maybe some hardware will implicitly only read a single byte of the word if configured to do so, that's unclear to me.

gramsay0, 2024-04-09:
Fair enough, maybe I'm misinterpreting what `source_data_size`/`dest_data_size` are actually for...

Looking at the NXP implementation I think `source_data_size` ends up configuring the `SSIZE` in the descriptor:
[image]
which impacts the number of reads required to reach `NBYTES`:
[image]
That seems to me like setting `source_data_size = 1` and `dest_data_size = 4` would mean the source data would be read 4 individual bytes before writing dest, **but the overall transfer size (number of bytes) is the same for source/dest**.

I had taken `source_data_size`/`dest_data_size` to mean you could do something like this:
> e.g. if you transfer 100 bytes into 100 32bit words, is that 100 or 400 bytes transferred?

but maybe that is not supported in the API, and I guess only possible on a subset of DMA controllers

teburd, 2024-04-10:
Even if you are doing the above it means the address adjust (stride) changes not that more  bytes are transferred.

gramsay0, 2024-04-10:
> Even if you are moving single bytes into 32 bit words like your example the same number of bytes are transferred.

If 100 bytes go into the DMA and 400 come out, I think it comes down to semantics on what "transferred" means.
i.e. do you take the source, dest, min or max size transferred

gramsay0, 2024-04-10:
Anyway, I think I can close this as me misinterpreting what `source_data_size`/`dest_data_size` mean.
Expanding the parameter comments with some more info might be helpful to future users.
Thanks for taking time to have a look into it