zephyrproject-rtos/zephyr #70605: scripts/west_commands/run_common.py : misleading error message in runners_yaml_path

louis-feller, 2024-03-22:
**Describe the bug**

`west flash` returns the following error message when runners.yaml doesn't exist in <application_build_directory>/zephyr/ :

>  FATAL ERROR: either a pristine build is needed, or board {board} doesn't support west flash/debug (no ZEPHYR_RUNNERS_YAML in CMake cache)

However, the cause of the error is not the absence of ZEPHYR_RUNNERS_YAML in CMake cache, but rather the absence of the runners.yaml file itself.
That is evident from the source code and from the reproducing procedure. Thus, the error message is misleading and should be changed, as it does not accurately describe what causes the error.

**To Reproduce**
Steps to reproduce the behavior:

Part 1 :  verify that the error isn't actually caused by the absence of  ZEPHYR_RUNNERS_YAML in CMakeCache :
1. `west build -b nucleo_wba55cg -p always samples/basic/blinky` (I used nucleo-WBA55CG as an example but this shouldn't be board-specific)
2. in build/CMakeCache.txt, look for the line starting with `ZEPHYR_RUNNERS_YAML` (line 580 in my case) and delete it.
4. `west flash --skip-rebuild` (use --skip-rebuild to prevent west from regenerating cache. This isn't necessary to reproduce but makes the bug more apparent as ZEPHYR_RUNNERS_YAML will still be unset after flashing).
5. No error, flashing works as intended

Part 2 : verify that no runners.yaml triggers the error even when ZEPHYR_RUNNERS_YAML is set :
1. `west build -b nucleo_wba55cg -p always samples/basic/blinky` (I used nucleo-WBA55CG as an example but this shouldn't be board-specific)
2. in build/zephyr : rename build/zephyr/runners.yaml to another name. For instance :  
`cd build/zephyr && mv runners.yaml runners`
3. Verify that ZEPHYR_RUNNERS_YAML is set in build/CMakeCache.txt
4. in build directory or application directory : `west flash`
5. >  -- west flash: rebuilding
ninja: no work to do.
FATAL ERROR: either a pristine build is needed, or board {board} doesn't support west flash/debug (no ZEPHYR_RUNNERS_YAML in CMake cache)

**Expected behavior**
When no runners.yaml file is present in the build system, the error message should accurately describe the problem, in line with what actually caused the error to happen in the code (trying to open a file that doesn't exist in this case).
A more accurate error message would be something like "No runners.yaml found in <build_dir>/zephyr . A pristine build might be needed.

**Impact**
Annoyance when tweaking with the build system/runners.

**Environment :**
 - OS: Ubuntu 22.04.4
 - Toolchain : Zephyr SDK 0.16.5-1
 - Zephyr 3.6.99 (commit SHA : b25eb36cb81917d529212c5a5c3d31ed56cd568d)

nordicjm, 2024-03-22:
I don't get it, you are manually altering the cache or removing a file, not really sure what you expect to happen? Don't manually edit the cache or remove the flash runner file?

louis-feller, 2024-03-22:
I'll add some context to clarify the issue : I am tweaking with the build system (with CMake) and adding extra images to be flashed alongside the zephyr image itself.

My problem is **not** that there is an error and that flashing doesn't work. This is expected.
My problem is that the **error message** I got points to a **missing cache variable** when the actual issue is a **missing file**.

Now, this is my first contribution and maybe the "bug" label is unfit for this issue. If so please let me know. I haven't found a very precise definition outlining what qualifies as a" bug" vs "enhancement" vs "feature" etc. If such a resource exists also please let me know.

tejlmand, 2024-03-22:
Thanks for raising this.
While I do know how annoying it is to be sent off track when debugging, then I have changed into `Enhancement` as the error message for sure can be improved.
A bug is more when something is broken in the system, which isn't the case here.

louis-feller, 2024-03-22:
Perfect, thank you.

PR #72237 (2024-06-28): west : runners : Fix error message when runners.yaml is not found.
Adjust error message so that it clearly states runners.yaml is missing from <build_dir>/zephyr, instead of referencing CMake cache variable ZEPHYR_RUNNERS_YAML, which is no longer used (since 3124c02987071b045b33bbfa2d12606b2ff0a757 ).
Also clean up that variable in CMake since it is no longer used (0 other references in entire tree).
Fixes #70605
commit 5b4c8945 (2024-06-28): west : runners : Fix error message when runners.yaml is not found.

Adjust error message so that it clearly states runners.yaml is
missing from <build_dir>/zephyr, instead of referencing CMake cache
variable ZEPHYR_RUNNERS_YAML, which is no longer used (since
3124c02987071b045b33bbfa2d12606b2ff0a757 ).
Also clean up that variable in CMake since it is no longer used
(0 other references in entire tree).
Fixes #70605
closed 2024-06-28 by nashif