zephyrproject-rtos/zephyr #70650: `sample.bluetooth.hci_usb.device_next` broken on stm32wb55

lopsided98, 2024-03-24:
**Describe the bug**

The `hci_usb` sample is non-functional when used with the new USB device stack on the STM32WB55. The command buffer sent to the HCI driver is malformed, with `buf->data == NULL` and `buf->len > buf->size`, causing the following error message to be printed: `bt_hci: Error sending to driver`.

I'm having some trouble following the data flow in the `udc_stm32` driver, but I traced at least as far as `HAL_PCD_DataOutStageCallback()`. In that function, `udc_buf_get()` returns a buffer with `len == size == 0` and `data == NULL`. The code then proceeds to call `net_buf_add(buf, rx_count)` on the buffer, where `rx_count > 0`. This results in the buffer with `len > size` and would cause a assertion failure with `CONFIG_ASSERT`. In practice, enabling `CONFIG_ASSERT` causes a crash earlier due to #61464.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Nucleo WB55 board
2. Add missing `NULL` argument to `sample_usbd_init_device()` call in `hci_usb` sample to fix build
3. `west build samples/bluetooth/hci_usb -DCONF_FILE=usbd_next_prj.conf --board nucleo_wb55rg`
4. `west flash`
5. Observe that the USB HCI device does not work

**Expected behavior**
`hci_usb` should be functional with the new USB stack

**Impact**
The old USB stack must be used to get `hci_usb` functionality on the STM32WB55.

**Logs and console output**

```
*** Booting Zephyr OS build v3.6.0-1547-g6c300c9c66e7 ***
[00:00:00.003,000] <inf> bt_hci_raw: Bluetooth enabled in RAW mode
[00:00:00.004,000] <err> usbd_core: Failed to set default address after bus reset
Bluetooth over USB sample
[00:00:00.463,000] <inf> bt_hci: Configuration enabled
[00:00:00.464,000] <err> bt_hci: Error sending to driver
```

**Environment:**
 - OS: Linux
 - Toolchain: GCC 13.2.0
 - Zephyr: v3.6.0-1547-g6c300c9c66e7

cc @loicpoulain

aescolar, 2024-03-26:
CC @erwango @FRASTM @ABOSTM  @gautierg-st @Desvauxm-st @GeorgeCGV

lopsided98, 2024-03-26:
After playing with the hci_usb (with the old USB stack) on the STM32WB55, it looks like the M0+ HCI firmware is far too buggy to be usable with Linux, so this is definitely a low priority.

erwango, 2024-07-30:
@asm5878 to provide an update

closed 2024-08-05 by erwango