zephyrproject-rtos/zephyr #70652: tests: libc: mimxrt1010_evk failed

hakehuang, 2024-03-25:
**Describe the bug**

tests/lib/c_lib/common/libraries.libc.common.picolibc.module
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc
tests/lib/c_lib/strerror/libraries.libc.strerror.newlib

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 mimxrt1010_evk
 - What have you tried to diagnose or workaround this issue?
 NA
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
thoes cases are not PASS since added

**To Reproduce**
Steps to reproduce the behavior:

west twister -p mimxrt1010_evk  -s tests/lib/c_lib/common/libraries.libc.common.picolibc.module

**Expected behavior**
test  PASS

**Impact**
c_lib supports

**Logs and console output**

```
*** Booting Zephyr OS build v3.6.0-1523-g37a1d28a36ae ***
Running TESTSUITE libc_cnd
===================================================================
START - test_cnd_broadcast_wait
E: ***** USAGE FAULT *****
E:   Unaligned memory access
E: r0/a1:  0x00000000  r1/a2:  0x20205104  r2/a3:  0xffffffff
E: r3/a4:  0xffffffff r12/ip:  0x0000c350 r14/lr:  0x6000fedf
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x6000ff16
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x202008a0 (unknown)
Caught system error -- reason 0 0
Fatal error was unexpected, aborting...
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-1523-g37a1d28a36ae

hakehuang, 2024-03-27:
all MXP RT series platform have such failures

decsny, 2024-06-04:
> tests/lib/c_lib/common/libraries.libc.common.picolibc.module
> tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc
> tests/lib/c_lib/strerror/libraries.libc.strerror.newlib

hi @hakehuang , I tried these test cases on the mimxrt1010_evk on the latest main 4fc6506a8af and did not see any test failures, can you double check

decsny, 2024-06-04:
@hakehuang please reopen if still seeing the issue

closed 2024-06-04 by decsny