zephyrproject-rtos/zephyr #70657: Problem with ADC conversion on stm32f3discovery board

Spago123, 2024-03-25:
**Bug**
Hello I have a problem with the adc conversion on stm32f3discovery board. When running the adc_read_dt() function I get an error. The error value is -22. So the program succesfully creates the adc_channal but fails to read the value. I am using the example code in zephyr/drivers/adc folder. I will provide you with the code and the overlay that I have added to support the stm32f3_disco board beacuse it was not given in the example folder. And also I have modified the stm32f3_disco.dts file so that I am measuring on the  gpio A1 that is on adc1  in2 channel.

**stm32f3_disco.overlay file**
/* Copyright (c) 2021 STMicroelectronics
   SPDX-License-Identifier: Apache-2.0 */

 / {
	zephyr,user {
		/* adjust channel number according to pinmux in board.dts */
		io-channels = <&adc1 1>;
	};
};

&adc1 {
	#address-cells = <1>;
	#size-cells = <0>;
	#status = "okay";

	channel@1 {
		reg = <1>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <16>;
	};
};

**Modified part of stm32f3_disco.dts file**
&adc1 {
	pinctrl-0 = <&adc1_in2_pa1>;
	pinctrl-names = "default";
	st,adc-clock-source = <SYNC>;
	st,adc-prescaler = <2>;
	status = "okay";
};

**src/main.c file**
/*
 * Copyright (c) 2020 Libre Solar Technologies GmbH
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <inttypes.h>
#include <stddef.h>
#include <stdint.h>

#include <zephyr/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/drivers/adc.h>
#include <zephyr/kernel.h>
#include <zephyr/sys/printk.h>
#include <zephyr/sys/util.h>

#if !DT_NODE_EXISTS(DT_PATH(zephyr_user)) || \
	!DT_NODE_HAS_PROP(DT_PATH(zephyr_user), io_channels)
#error "No suitable devicetree overlay specified"
#endif

#define DT_SPEC_AND_COMMA(node_id, prop, idx) \
	ADC_DT_SPEC_GET_BY_IDX(node_id, idx),

/* Data of ADC io-channels specified in devicetree. */
static const struct adc_dt_spec adc_channels[] = {
	DT_FOREACH_PROP_ELEM(DT_PATH(zephyr_user), io_channels,
			     DT_SPEC_AND_COMMA)
};

int main(void)
{
	int err;
	uint32_t count = 0;
	uint16_t buf;
	struct adc_sequence sequence = {
		.buffer = &buf,
		/* buffer size in bytes, not number of samples */
		.buffer_size = sizeof(buf),
	};

	/* Configure channels individually prior to sampling. */
	for (size_t i = 0U; i < ARRAY_SIZE(adc_channels); i++) {
		if (!adc_is_ready_dt(&adc_channels[i])) {
			printk("ADC controller device %s not ready\n", adc_channels[i].dev->name);
			return 0;
		}

		err = adc_channel_setup_dt(&adc_channels[i]);
		if (err < 0) {
			printk("Could not setup channel #%d (%d)\n", i, err);
			return 0;
		}
	}

	while (1) {
		printk("ADC reading[%u]:\n", count++);
		for (size_t i = 0U; i < ARRAY_SIZE(adc_channels); i++) {
			int32_t val_mv;

			printk("- %s, channel %d: ",
			       adc_channels[i].dev->name,
			       adc_channels[i].channel_id);

			(void)adc_sequence_init_dt(&adc_channels[i], &sequence);

			err = adc_read_dt(&adc_channels[i], &sequence);
			if (err < 0) {
				printk("Could not read (%d)\n", err);
				continue;
			}

			/*
			 * If using differential mode, the 16 bit value
			 * in the ADC sample buffer should be a signed 2's
			 * complement value.
			 */
			if (adc_channels[i].channel_cfg.differential) {
				val_mv = (int32_t)((int16_t)buf);
			} else {
				val_mv = (int32_t)buf;
			}
			printk("%"PRId32, val_mv);
			err = adc_raw_to_millivolts_dt(&adc_channels[i],
						       &val_mv);
			/* conversion to mV may not be supported, skip if not */
			if (err < 0) {
				printk(" (value in mV not available)\n");
			} else {
				printk(" = %"PRId32" mV\n", val_mv);
			}
		}

		k_sleep(K_MSEC(1000));
	}
	return 0;
}

**Note**
Other files are the same as in the example folder shown in zephyr/drivers/adc. If you could provide me with a solution I would be grateful.

gautierg-st, 2024-03-25:
Hello,
You're trying to read the channel 2 of the ADC (pin adc1_in2_pa1), so your overlay should reflect that:
```
/ {
	zephyr,user {
		io-channels = <&adc1 2>;
	};
};

&adc1 {
	channel@2 {
		reg = <2>;
		...
	}
	...
}
```

Spago123, 2024-03-25:
> Hello, You're trying to read the channel 2 of the ADC (pin adc1_in2_pa1), so your overlay should reflect that:
> 
> ```
> / {
> 	zephyr,user {
> 		io-channels = <&adc1 2>;
> 	};
> };
> 
> &adc1 {
> 	channel@2 {
> 		reg = <2>;
> 		...
> 	}
> 	...
> }
> ```

Thank you a lot gautierg-st. Everything is working now.

closed 2024-03-25 by erwango