zephyrproject-rtos/zephyr #70666: SRAM2 is being unused

AlbertPJoshi, 2024-03-25:
I am working on an NXPS32 based MCU, while building the MCUboot code the RAM area is overflowed. During the investigation i found out that in the current set up(based on the post build prints and linker scripts, .map file) zephyr not using the SRAM_U(NXPs32k have two separate RAM regions called SRAM_U and SRAM_L it is not continuous memory ).

So in order to use this area i have tried modifying the linker using CONFIG_CODE_DATA_RELOCATION, but even though adding this  and some proposed changes on the CMakeList.txt i was not able to use the SRAM_U area.

Is there any way to combine these two RAM areas(it is not continuous as i mentioned).?
Or what is the proper way to modify the linker in order to use the SRAM_U?

**To Reproduce**
Steps to reproduce the behavior:
1. Build MCUboot with all the encryption on a 64kb RAM MCU( having two separate RAM partitions)

AlbertPJoshi, 2024-03-27:
Hi @manuargue 
I am using UCANS32K1SIC (S32K146) and also tried S32K144_q100_evaluation board (S32k144), but my application build is getting failed by limited size of the SRAM region of 64k in s32K146 and 32K in S32K144 respectively, 

since only either SRAM_U or SRAM_L could be selected using chosen node **zephyr,sram** 
Is there any way around this?

 I wanted a way to combine/use he SRAM_U and SRAM_L region for my zephyr application.

AlbertPJoshi, 2024-03-27:
I am trying to fit MCUBoot in s32k144(32K sram_l and 28K sram_l), But the usable reqion of sram_l is not enough for this, we tried this method by extending the sram_u region to 64k, but I assume mcuboot uses the area in between the boundary for image swapping (Using Ram), without scratch partition, and fails in between switching the application image.It maybe due to the fore mentioned burst access issue.

Also without additional ram I am unable to fit Mcuboot with encryption and other features.
So, how can i guarantee that my application avoids burst access across the boundary?

AlbertPJoshi, 2024-03-28:
Thank you for the reply,

Can you help/guide me on adding a custom linker script that achieves this behaviour?
 I already tried to tinker with this but could achieve this behaviour.

manuargue, 2024-10-15:
As detailed above, unaligned and burst accesses cannot occur across the banks boundary. Current implementation aligns with similar devices such as Kinetis which present the same hw limitation. User can use a custom linker script to adapt to their needs the default configuration.

closed 2024-10-15 by manuargue