zephyrproject-rtos/zephyr #70674: drivers: video: STM32 Digital camera interface (DCMI) driver

CharlesDias, 2024-03-25:
**Is your feature request related to a problem? Please describe.**
I'm trying to use the OV2640 camera module on the NUCLEO-H743ZI2 board. However, the video driver for the DCMI peripheral doesn't exist (the driver for OV2640 does exist).  

**Describe the solution you'd like**
Development of the DCMI video driver to capture the images from a camera module.

**Additional context**
Initially, I created a mock DCMI video driver version to generate two mock test images (color bar and gradient images). To validate this mock version, I change the `sample/subsys/video/tcpserversink` to be able to run on NUCLEO-H743ZI2. Due the memory limitation, I reduce the video resolution and the Ethernet TX buffers. This mock version worked as expected. See the video.

I read the AN5020 document from ST, which helped me understand the DCMI peripheral and its requirements. After that, I started to create the real DCMI video driver, but I am stuck right now with the DMA configuration for DCMI.

This is my Zephyr's fork with my current DCMI driver version. I can continue working on this driver if someone can help me to unblock. ;-)

https://github.com/CharlesDias/zephyr/tree/driver_stm32_dcmi/drivers/video

/cc @kartben @erwango @FRASTM 

https://github.com/zephyrproject-rtos/zephyr/assets/27859307/cb8a5122-001c-44cc-a1cf-74f5078ea05b

adityaganesh2k, 2024-04-10:
Hey @CharlesDias I was going through your changes to add the DCMI driver . Is there a possibility that we can add camera support also along with video support in this PR itself . I would like to contribute to that as I am also currently working on DCMI support for OV2640 by myself

CharlesDias, 2024-04-10:
Hi, @adityaganesh2k. Glad to know there's someone working on this too! I can open the PR as a draft to get help from you and others. What do you think?

adityaganesh, 2024-04-10:
Yeah that would be great i am working with a Weact Studio board based on stm32h750vbt6 . Link for the board:-https://evelta.com/weact-studio-stm32h750-development-board/. 
I actually made the same changes as your commit on a branch of mine but my branch is forked from v3.6 of zephyr where iam getting a compilation error . My board files and the changes to add dcmi looks fine to me . If you could share me your email id we can correspond there

CharlesDias, 2024-04-13:
Hi, @adityaganesh.

Sure. My e-mail is charlesdias.cd@outlook.com. Also, if you prefer, send me a message on LinkedIn 
https://www.linkedin.com/in/charles-dias-m-sc-6ab50527/

I opened the PR https://github.com/zephyrproject-rtos/zephyr/pull/71462 related to the STM32 DCMI driver. I tested with the Weact Studio STM32H743 board and worked fine.

erwango, 2024-06-27:
@CharlesDias Do you think this point can be closed ?

CharlesDias, 2024-06-27:
> @CharlesDias Do you think this point can be closed ?

Hi, @erwango. The sample application using the DCMI driver hasn't been merged yet, PR #71463 . Is this a reason not to close the issue? What do you think?

erwango, 2024-06-27:
ok, sure. Maybe you want to update the TODO list above then.

avolmat-st, 2026-01-19:
@CharlesDias I think all listed points are done and merged now so can we close this issue ?