zephyrproject-rtos/zephyr #70682: `counter_set_channel_alarm` on RTC not working with `CONFIG_PM` and sleep

laurin, 2024-03-25:
**Description of the bug**
I am using Zephyr with a `nucleo_l4r5zi` board.
I want to use the counter API with the rtc to schedule an alarm. This will later be used to wake up the device from standby.
However, if I enable `CONFIG_PM` and call `k_msleep` before calling `counter_set_channel_alarm`, the assigned callback function will not be called.

The behaviour is also weird because the `printk` below the `k_msleep` also does not seem to be executed, even though it is located earlier than the line that sets the counter alarm.

I would guess the app directly configuring the RTC somehow messes with the power management scheduling zephyr does.
However, I have not found a way to do this another way, since only the RTC can be used to wake up the device from standby, so I cannot use another timer.
But I have not really found much regarding this topic with Zephyr, so maybe I am using this totally wrong and someone can point me into the right direction. :)

**To Reproduce**

```c
// prj.conf

CONFIG_COUNTER=y
CONFIG_PM=y
CONFIG_IDLE_STACK_SIZE=1024
```

```c
// main.c

#include <zephyr/kernel.h>
#include <zephyr/drivers/counter.h>

#define WAIT_TIME_SEC 5

static void counter_cb(const struct device *counter_dev,
				      uint8_t chan_id, uint32_t ticks,
				      void *user_data)
{
	printk("counter callback called\n");
}

int main(void)
{
	int ret;
	const struct device *dev = DEVICE_DT_GET(DT_NODELABEL(rtc));
	struct counter_alarm_cfg alarm_cfg = { 0 };

	if (!device_is_ready(dev)) {
		printk("Counter device not ready\n");
		return 0;
	}

	ret = counter_start(dev);
	if (ret < 0) {
		printk("Could not start counter (%d)\n", ret);
		return 0;
	}

	alarm_cfg.ticks = counter_us_to_ticks(dev, WAIT_TIME_SEC * USEC_PER_SEC);
	alarm_cfg.callback = counter_cb;
	
	printk("Hello world\n");

	// counter_cb is only called if this is commented out
	k_msleep(2000);

	printk("Setting counter\n");

	ret = counter_set_channel_alarm(dev, 0, &alarm_cfg);
	if (ret < 0) {
		printk("Could not set counter channel alarm (%d)\n", ret);
		return 0;
	}

	printk("Counter set to %d seconds\n", WAIT_TIME_SEC);

	k_msleep(10000);

	return 0;
}
```

`west build -b nucleo_l4r5zi -p`

**Expected behavior**
The counter callback is called even if the system sleeps before setting the counter.
Expected log:
```log
Hello world
-- 2s wait --
Setting counter
Counter set to 5 seconds
-- 5s wait --
counter callback called
```

**Impact**
Showstopper if I need the device to wake up after it was put to sleep.

**Logs and console output**

Log output when not calling `k_msleep(2000)`
```
Hello world
Setting counter
Counter set to 5 seconds
-- 5s wait --
counter callback called
```

Log output when calling `k_msleep(2000)`
```
Hello world
```

**Environment:**
 - OS: Windows
 - Toolchain: zephyr-sdk-0.16.3
 - Zephyr v3.6

**Additional context**
As I wrote above, I want to use this to wake up the device from standby. This already works, if there is no sleep statement before the counter configuration.

laurin, 2024-03-25:
This sounds very related to #68673, but unfortunately applying the changes introduced in #70078 did not fix my issues

laurin, 2024-03-25:
Ok, due to unfortunate configuration while testing, I did not notice that #70078 does indeed resolve my issue.
I did not enable the LPTIM in the device tree overlay, so the device simply did not wake up after sleeping.

closed 2024-03-25 by laurin

zafersn, 2024-07-14:
Hi @ajarmouni-st,  Could you please share your thought here?  I built my rtc test app on [the sample power blinky app](https://github.com/zephyrproject-rtos/zephyr/tree/v3.6-branch/samples/boards/stm32/power_mgmt/blinky) and ran on the nucleo_u575zi board, and RTC calback works till the device goes into sleep mode. In sleep mode, everything works until the rtc timer expires, and then the app stuck. (I can see current consumption differences when the timer expires)

current consumption before rtc time expires:

[image]

after rtc time expires: and app stuck no more process.

[image]

dts output file:

[zephyr.dts.txt](https://github.com/user-attachments/files/16228484/zephyr.dts.txt)

main app:

```
/*
 * Copyright (c) 2021 Linaro Limited
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/drivers/gpio.h>
#include <zephyr/sys/printk.h>
#include <zephyr/drivers/counter.h>

#define SLEEP_TIME_MS   2000
#define ALARM_PERIOD_SEC ( 5)

static const struct gpio_dt_spec led =
	GPIO_DT_SPEC_GET(DT_ALIAS(led0), gpios);

const struct device *rtc_dev;
struct counter_alarm_cfg alarm_cfg;

static void test_counter_interrupt_fn(const struct device *counter_dev,
				      uint8_t chan_id, uint32_t ticks,
				      void *user_data)
{
	printk("!!! Alarm !!!\n");
}

int configure_rtc(void)
{
    rtc_dev = DEVICE_DT_GET(DT_NODELABEL(rtc));
  	if (!device_is_ready(rtc_dev)) {
		printk("device not ready.\n");
		return 0;
	}
	counter_start(rtc_dev);
    alarm_cfg.flags = 0;
    alarm_cfg.ticks = counter_us_to_ticks(rtc_dev, ALARM_PERIOD_SEC * USEC_PER_SEC);
    alarm_cfg.callback = test_counter_interrupt_fn;
    alarm_cfg.user_data =  &alarm_cfg;

    if (counter_set_channel_alarm(rtc_dev, 0, &alarm_cfg)) {
        printk("Unable to set RTC alarm\n");
    } else {
        printk("RTC alarm set for 24 hours\n");
    }
	printk("Set alarm in %u sec (%u ticks)\n",
	       (uint32_t)(counter_ticks_to_us(rtc_dev,
					   alarm_cfg.ticks) / USEC_PER_SEC),
	       alarm_cfg.ticks);
	return 0;
}
int main(void)
{
	bool led_is_on = true;

	__ASSERT_NO_MSG(gpio_is_ready_dt(&led));
	configure_rtc();
	printk("Device ready\n");

	while (true) {
		printk("led is %d\n", led_is_on);
		gpio_pin_configure_dt(&led, GPIO_OUTPUT_ACTIVE);
		gpio_pin_set(led.port, led.pin, (int)led_is_on);
		if (led_is_on == false) {
			/* Release resource to release device clock */
			gpio_pin_configure(led.port, led.pin, GPIO_DISCONNECTED);
		}
		k_msleep(SLEEP_TIME_MS);
		if (led_is_on == true) {
			/* Release resource to release device clock */
			gpio_pin_configure(led.port, led.pin, GPIO_DISCONNECTED);
		}
		led_is_on = !led_is_on;
	}
	return 0;
}

```
prj.conf:

```
CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
CONFIG_PM_DEVICE_RUNTIME_EXCLUSIVE=n
CONFIG_ASSERT=y

CONFIG_COUNTER=y
CONFIG_COUNTER_RTC_STM32=y
# CONFIG_STM32_WKUP_PINS=y
CONFIG_COUNTER_RTC_STM32_CLOCK_LSE=y
CONFIG_COUNTER_TIMER_STM32=n

CONFIG_POWER_SUPPLY_DIRECT_SMPS=y
```
just in case I have attached the sample app with overlay.
[blinky_st_pm - Copy.zip](https://github.com/user-attachments/files/16228467/blinky_st_pm.-.Copy.zip)