zephyrproject-rtos/zephyr #70693: Bluetooth: Audio: Broadcast assistant should only set `BT_BAP_BASS_PA_REQ_SYNC_PAST` if it has a sync

Thalley, 2024-03-25:
**Is your enhancement proposal related to a problem? Please describe.**
The `bt_bap_broadcast_assistant_add_src` function will automatically set `pa_sync = BT_BAP_BASS_PA_REQ_SYNC_PAST` if `past_available` returns true. `past_available` is however only based on features, and not on whether we have an actual sync. Since PAST only works if we have a sync, then `past_available` should take that into account, or there needs to be an additional check. 

Alternatively the `pa_sync` value should be provided by the application. 

**Describe the solution you'd like**
The `bt_bap_broadcast_assistant_add_src` (and possible other places), should not set `pa_sync = BT_BAP_BASS_PA_REQ_SYNC_PAST` unless the device has a PA sync to the source being added. 

**Describe alternatives you've considered**
Alternatively leave it up to the application and modify `pa_sync` to be a non-binary value in `struct bt_bap_broadcast_assistant_add_src_param`. 

**Additional context**
N/A

aescolar, 2024-03-26:
@Thalley when submitting a bug report for your own subsystem, please also set the priority to it, as you are the best to know what it should be.

Thalley, 2024-03-26:
> @Thalley when submitting a bug report for your own subsystem, please also set the priority to it, as you are the best to know what it should be.

Sure - We usually do that every week on Tuesday as a group, but today's meeting was cancelled. I may be a maintainer, but not a dictator :D

Thalley, 2024-05-09:
This bug report is wrong: The broadcast assistant also takes the PA sync into account by doing at lookup with bt_le_per_adv_sync_lookup_addr: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/bap_broadcast_assistant.c#L146-L157

closed 2024-05-09 by Thalley