zephyrproject-rtos/zephyr #70696: PPP: k_yield() in ppp_send_flush

andreasmueller, 2024-03-25:
## Introduction

We noticed that PPP blocks other threads for a long time when sending a packet to the host over UART (e.g. ~50 ms when sending a 500 byte packet at 115200 bps).

### Problem description

When PPP sends a packet over UART, the `tx_q` thread does not yield until all data is sent. When the packet is large, this can take a while and other threads – including higher priority `tx_q` threads – are blocked during this time (`CONFIG_NET_TC_THREAD_PREEMPTIVE` is not set).

### Proposed change

Adding a `k_yield()` to `ppp_send_flush()` allows higher-priority threads to run.

## Detailed RFC

I have tested the following change and this solves the issue for us:

```c
diff --git a/drivers/net/ppp.c b/drivers/net/ppp.c
index 67ebc0f3b0c..dcf62f53c4b 100644
--- a/drivers/net/ppp.c
+++ b/drivers/net/ppp.c
@@ -380,6 +380,15 @@ static int ppp_send_flush(struct ppp_driver_context *ppp, int off)
        } else {
                while (off--) {
                        uart_poll_out(ppp->dev, *buf++);
+                       /* yield control every N bytes so that higher-priority threads can run
+                        * (note: N=28 should be about 2 ms at 115200 bps)
+                        */
+                       static uint8_t counter = 0;
+                       counter++;
+                       if (counter % 28 == 0) {
+                               counter = 0;
+                               k_yield();
+                       }
                }
        }
``` 

### Proposed change (Detailed)

With this change, `ppp_send_flush()` will yield after every N bytes, allowing other threads to run. This means the UART transfer may be interrupted; this should not be an issue however.

### Dependencies

–

### Concerns and Unresolved Questions

Questions:

- is the current behavior considered a bug?
- should I upstream this change?
- if yes, should it be behind a KConfig & disabled by default?
- if yes, should there be a KConfig determining after how many bytes to yield?

## Alternatives

- `CONFIG_NET_TC_THREAD_PREEMPTIVE` would be an alternative, but is not viable in our case as another network stack running on the same device does not support it.

rlubos, 2024-03-26:
Yielding will only work with threads at the same or higher priority level, lower prior threads will still not be able to run in such case.
Wouldn't it be better to switch to interrupt driver or asynchronous UART API to mitigate the problem?
https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html#universal-asynchronous-receiver-transmitter-uart

rlubos, 2024-03-26:
> CONFIG_UART_INTERRUPT_DRIVEN is active, but does not solve the issue. 

Changing the config doesn't affect `uart_poll_out()` behavior, it's still a blocking function. The PPP driver would need to be rewritten to use interrupt-driver API: https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html#uart-interrupt-api

rlubos, 2024-03-27:
> OK, that makes sense. Rewriting the PPP driver sounds like a bigger change though. Given that my change above works fine for us based on my testing so far, I'll stick with that for now.

My personal opinion is that we should aim to provide solutions that would help in either case (lower prior threads as well), instead of patching the driver with fixes that would only help in certain cases. So switching to interrupt-driver UART APIs would be a wat to go with this.

If you don't have time to pursue this, maybe we could just convert this to an enhancement issue (describing preferred solution) and someone may pick it up in the future?

andrmuel, 2024-03-28:
That sounds good to me. Since my use-case works with the solution using `k_yield()`  I'd alternatively also be fine with just closing this issue and if someone has a use case that requires a solution that allows lower-prio threads to run as well, a new ticket can be opened.