zephyrproject-rtos/zephyr #70699: STM32 RTC driver null pointers

krzysztofgawrys, 2024-03-25:
**Description**
Target Platform: STM32 (H723VG but possibly any STM32 MCU with hardware RTC)
Build path: samples/drivers/counter/alarm modified to use HW RTC instead of Timer

Added RTC to DTS:
```
&rtc {
	clocks = <&rcc STM32_CLOCK_BUS_APB4 0x00010000>,
	         <&rcc STM32_SRC_LSE RTC_SEL(1)>;
	status = "okay";
};
```
and RTC to prj.conf:
```
CONFIG_RTC=y
CONFIG_RTC_ALARM=y
```

Project builds without errors, but kernel will crash because it is lacking:
```
CONFIG_COUNTER_RTC_STM32=y
CONFIG_COUNTER=y
```

RTC driver api is in corrupted state, some function have null pointers, some have completely incorrect addresses:
[image]

**Expected behavior**
Build assertion error.

**Environment**
OS: Ubuntu on WSL2
Toolchain: 
zephyr v3.6.0-214
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

ajarmouni-st, 2024-04-08:
> Project builds without errors, but kernel will crash because it is lacking:
> ```
> CONFIG_COUNTER_RTC_STM32=y
> CONFIG_COUNTER=y
> ```
> RTC driver api is in corrupted state, some functions have null pointers, some have completely incorrect addresses:

Hi @krzysztofgawrys, that's the expected behavior, RTC_ALARM is implemented in [counter_ll_stm32_rtc.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/counter/counter_ll_stm32_rtc.c) and it's not yet available in [rtc_ll_stm32.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/rtc/rtc_ll_stm32.c#L406-L407) as it is a recent driver.

krzysztofgawrys, 2024-04-08:
> In order to use RTC alarm in `samples/drivers/counter/alarm`, it should be run with `CONFIG_COUNTER_RTC_STM32=y`and `CONFIG_COUNTER_TIMER_STM32=n`.

Hi @ajarmouni-st, i agree with that, however error during build would be nice to have in this case, I've spend several hours debugging my problem and i guess I'm not the first one nor the last one.

erwango, 2024-04-08:
> Hi @ajarmouni-st, i agree with that, however error during build would be nice to have in this case, I've spend several hours debugging my problem and i guess I'm not the first one nor the last one.

That's a valid point. We need to check if it is possible to generate an error on misconfigurations

ajarmouni-st, 2024-04-09:
@krzysztofgawrys Do you think that this [commit](https://github.com/zephyrproject-rtos/zephyr/pull/71208/commits/956c936f9a0a3c3c6631aa50b7e9ff81ee256115) somehow addresses the lack of clarity?

nashif, 2024-04-09:
Please do not remove bug label without add a different label with reasoning.

laurin, 2024-04-10:
Hey, do you have a suggestion on what to do if I need to use the RTC for both alarms and getting the datetime?
Currently, I have to enable `CONFIG_RTC_STM32` if I want to get/set the dateime and `CONFIG_COUNTER_RTC_STM32` if I want to use the alarms.
If I enable both, the build will produce a `multiple definition`-error for the rtc dts-device, because both drivers want to create a device object for it.
So currently I have to choose which one I want. 😁
I can you open another issue if you think that this does not belong in here.

erwango, 2024-04-10:
> I can you open another issue if you think that this does not belong in here.

Yes, please open a dedicated issue

krzysztofgawrys, 2024-04-15:
> @krzysztofgawrys Do you think that this [commit](https://github.com/zephyrproject-rtos/zephyr/pull/71208/commits/956c936f9a0a3c3c6631aa50b7e9ff81ee256115) somehow addresses the lack of clarity?

Yes, this is clear.

closed 2024-06-26 by nashif