zephyrproject-rtos/zephyr #70725: build warning on  cast removes address space '__cache' of expression

kv2019i, 2024-03-26:
**Describe the bug**
Sparse warning on use of __cache attribute:

https://github.com/thesofproject/sof/actions/runs/8422916305/job/23063323303?pr=8981

 /zep_workspace/zephyr/soc/intel/intel_adsp/ace/power.c:46:12: warning: cast removes address space '__cache' of expression
/zep_workspace/zephyr/soc/intel/intel_adsp/ace/power.c:48:9: warning: incorrect type in argument 1 (different address spaces)
Regression in 

**To Reproduce**
https://github.com/thesofproject/sof/actions/runs/8422916305/job/23063323303?pr=8981

**Expected behavior**
No sparse warnings on __cache attribute usage.

**Impact**
Build warning

**Logs and console output**
https://github.com/thesofproject/sof/actions/runs/8422916305/job/23063323303?pr=8981

**Environment (please complete the following information):**
 - https://github.com/thesofproject/sof/actions/runs/8422916305/job/23063323303?pr=8981
 
**Additional context**
Regression in commit 6b9d01f995c72d1c747a98dbc866d052a0f29cb0  . Seems cache_data_flush_range() does not have cache attributes.

nashif, 2024-03-27:
@ceolin  can you please take a look?

ceolin, 2024-03-28:
> @nashif , @ceolin can we add this check to either [`main`/.github](https://github.com/zephyrproject-rtos/zephyr/tree/main/.github?rgh-link-date=2024-03-28T16%3A02%3A06Z) or Internal Intel CI? Either would shorten the feedback loop considerably.
> 
> The required dependencies are small:
> 
> [thesofproject/sof@0698fa334ae16b7709f08](https://github.com/thesofproject/sof/commit/0698fa334ae16b7709f08)

I think so ... also we probably need to have our sparse fork. The way it is on Zephyr right now assumes that you can get any sparse (from your distro for example) and it will work

marc-hb, 2024-03-29:
> > [thesofproject/sof@0698fa334ae16b7709f08](https://github.com/thesofproject/sof/commit/0698fa334ae16b7709f08)

> I think so ... also we probably need to have our sparse fork. 

- Compile a reasonably recent version from the sparse main branch: yes. This is trivial, see link above.
- "Fork": no, not needed any more. We worked with the sparse maintainer and sparse now works with Zephyr out of the box.

All what's required is at the link above, it's pretty short.

PR #70806 (2024-03-29): intel_adsp: ace: Fix sparse error
Fixes the following errors when sparse (SCA) is enabled:

soc/intel/intel_adsp/ace/power.c:46:12: warning:
    cast removes address space '__cache' of expression
/soc/intel/intel_adsp/ace/power.c:48:9: warning:
    incorrect type in argument 1 (different address spaces)

Fixes #70725
commit 688fbb53 (2024-03-29): intel_adsp: ace: Fix sparse error

Fixes the following errors when sparse (SCA) is enabled:

soc/intel/intel_adsp/ace/power.c:46:12: warning:
    cast removes address space '__cache' of expression
/soc/intel/intel_adsp/ace/power.c:48:9: warning:
    incorrect type in argument 1 (different address spaces)

Fixes #70725
closed 2024-03-29 by dleach02