zephyrproject-rtos/zephyr #70735: Bluetooth: Optimize data passing between Host and Controller for ISO

KAGA164, 2024-03-26:
**Is your enhancement proposal related to a problem? Please describe.**

The Bluetooth host functions that send data are asynchronous and require context switching before data packet send over HCI to the controller. This adds a lot of latency especially when data is sent over ISO channels with timestamp by using the `bt_iso_chan_send_ts`.
Some BLE controllers have a possibility to read timestamp of the last SDU to synchronize data generation in an application. Due to asynchronous nature of the `bt_iso_chan_send_ts` function timestamp data cannot be obtained just after calling send function because context needs to be switched to HCI Tx thread and this time is not deterministic, so the only place when we can get timestamp is sent callback. However the latency between calling send function and sent callback reception with reading timestamp is big, around ~1.9 ms on nRF5340 device.This value is to big for new protocols being developed under BLE SIG which assume 1ms data latency.

**Describe the solution you'd like**

The fastest solution is to add synchronous version of the  `bt_iso_chan_send_ts` and `bt_iso_chan_send` when data would be delivered to the controller before returning from these functions. Then timestamp of this packet could be read immediately and  this would remove a delay introduced by waiting for sent callback.

The new functions could be following:

- `bt_iso_chan_send_ts_sync`
- `bt_iso_chan_send_sync`

**Describe alternatives you've considered**
Alternative solution could be optimization of the communication between BLE controller and Host by reducing context switching.

alwa-nordic, 2024-04-02:
I'm struggling to understand what core of problem is. There is a lot happening in your problem description. Maybe there is more than one issue here?

I'll start with the title; "Optimize data passing between Host and Controller for ISO". Is the problem that it takes too long between a call to `bt_iso_chan_send_ts_sync` and when the data has been sent over HCI?

Thalley, 2024-12-21:
Blocking the call when using e.g. `bt_iso_chan_send_ts_sync` until the SDU has been sent on air is not a good solution IMO. Besides requiring a completely new TX path in the host, there are also the following issues with that approach:

1. If the ISO interval is large (e.g. 2 seconds) then this may block for too long. We could possibly prohibit using `bt_iso_chan_send_ts_sync` in specific threads. 
2. HCI has no time requirements, so even when you use `bt_iso_chan_send_ts_sync` the delay may large. 

This request seem to expect specific behavior of the controller, but the host is (or triyng to be) controller agnostic, and thus I don't think having a blocking TX function will really fix any issues. 

Additionally you refer to reading the SDU timestamp - Presumably by using `bt_iso_chan_get_tx_sync`. While this is a blocking call, it also has no timing requirements (HCI), and the implement of this command in the controller is, as of Core 6.0, still a bit ambiguous. That was also discussed in https://github.com/zephyrproject-rtos/zephyr/pull/83139. 

I suggest to close this enhancement with the reasoning that no guarantees can be applied for timing of ISO data over HCI. It is best effort, and may be significantly different depending on the controller used.

Thalley, 2025-07-16:
Closing this since there's been no response from the reporter. @KAGA164 please comment if you still think this should be added

closed 2025-07-16 by Thalley