zephyrproject-rtos/zephyr #70764: Error with image confirmation for non-Direct XIP DFU

tony-le-24, 2024-03-26:
**Describe the bug**
I found the issue that indicates `boot_write_img_confirmed returned 4` when confirming a non-Direct XIP DFU with MCUBoot, and reverts back to the original image. It did not happen before this commit https://github.com/zephyrproject-rtos/zephyr/commit/24192740730cd3021114a2e32283443ec18a8ab7. The commit seems to have broken non-Direct XIP DFUs. @besmarsh since you're the author of that commit.

**To Reproduce**
N/A

**Expected behavior**
non-Direct XIP DFU should be confirmed and updated.

**Impact**
N/A

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA: 0051731a41fa2c9057f360dc9b819e47b2484be5

**Additional context**
N/A

nordicjm, 2024-03-27:
Cannot replicate issue using `smp_svr`

besmarsh, 2024-03-27:
Sorry to hear you're having an issue after my change! I've re-tested non-Direct XIP DFU using MCUboot with a custom application today and it works for me. Can you provide any details on the application you're using where you see the problem?

tony-le-24, 2024-03-27:
Hi, thank you for your reply. I've just checked and found that someone from our group has been working on fixing the same issue.  I guess I don't need to duplicate the work. Have a good day.

nordicjm, 2024-03-28:
Not an upstream issue

closed 2024-03-28 by nordicjm