zephyrproject-rtos/zephyr #70777: I2S MCUX on IMXRT1176 causes DMA error when TX and RX are used simultanously

RadekPolyend, 2024-03-27:
I've been testing I2S MCUX on IMXRT1176 and come across a problem with using TX and RX channels at the same time(when used separately there is no problem). Proicessor seem to get stuck in handling dma_mcux_edma_error_irq_handler. When looked into dma registers I found out that at some point the TX channel (DMA 0 channel) got the DADDR changed into invalid but when I set the watchpoint on the address there is no break therefore it wasn't changed explicitly.

(gdb) c
Continuing.

Hardware watchpoint 4: *0x40071010

Old value = 1077952544
New value = 0
EDMA_TcdReset (tcd=tcd@entry=0x40071000) at /home/radekp/projects/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/edma/fsl_edma.c:459
459	    tcd->DLAST_SGA = 0U;
(gdb) x/16w 0x40071000
0x40071000:	0x00000000	0x00000000	0x00000000	0x00000000
0x40071010:	0x00000000	0x00000000	0x80000180	0x0372000a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa5894	0x006c0002	0x80000200	0x03720012
(gdb) c
Continuing.

Hardware watchpoint 4: *0x40071010

Old value = 0
New value = 1077952544
0x3003ce9c in EDMA_InstallTCD (base=<optimized out>, channel=<optimized out>, tcd=0x80000140 <tcdpool>) at /home/radekp/projects/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/edma/fsl_edma.c:101
101	    base->TCD[channel].DOFF          = tcd->DOFF;
(gdb) x/16w 0x40071000
0x40071000:	0x80aa6734	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x00000000	0x00000000	0x00000000
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa5894	0x006c0002	0x80000200	0x03720012
(gdb) c
Continuing.

Breakpoint 1, dma_mcux_edma_reload (dev=0x3004c4b0 <__device_dts_ord_486>, channel=0, src=2158649424, dst=1077952544, size=1764) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:476
476		struct call_back *data = DEV_CHANNEL_DATA(dev, channel);
(gdb) x/16w 0x40071000
0x40071000:	0x80aa6734	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x03720000	0x80000160	0x0372000a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa5894	0x006c0002	0x80000200	0x03720012
(gdb) c
Continuing.

Breakpoint 1, dma_mcux_edma_reload (dev=0x3004c4b0 <__device_dts_ord_486>, channel=1, src=1077952672, dst=2158645896, size=1764) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:476
476		struct call_back *data = DEV_CHANNEL_DATA(dev, channel);
(gdb) x/16w 0x40071000
0x40071000:	0x80aa67a2	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x033b0000	0x80000160	0x03720012
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa596c	0x03720002	0x800001e0	0x0372000a
(gdb) c
Continuing.

Breakpoint 1, dma_mcux_edma_reload (dev=0x3004c4b0 <__device_dts_ord_486>, channel=1, src=1077952672, dst=2158644132, size=1764) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:476
476		struct call_back *data = DEV_CHANNEL_DATA(dev, channel);
(gdb) x/16w 0x40071000
0x40071000:	0x80aa67b2	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x03330000	0x80000160	0x03720012
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa596c	0x03720002	0x800001e0	0x03720012
(gdb) c
Continuing.

Breakpoint 2, dma_mcux_edma_irq_handler (dev=0x3004c4b0 <__device_dts_ord_486>, channel=0) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:191
191		uint32_t hw_channel = dma_mcux_edma_add_channel_gap(dev, channel);
(gdb) x/16w 0x40071000
0x40071000:	0x80aa605a	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x036d0000	0x80000180	0x0372000a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa5f7c	0x006a0002	0x800001e0	0x03720012
(gdb) c
Continuing.

Breakpoint 2, dma_mcux_edma_irq_handler (dev=0x3004c4b0 <__device_dts_ord_486>, channel=0) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:195
195			LOG_DBG("IRQ OCCURRED");
(gdb) x/16w 0x40071000
0x40071000:	0x80aa605e	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x036b0000	0x80000180	0x0372000a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa5f80	0x00680002	0x800001e0	0x03720012
(gdb) c
Continuing.

Breakpoint 2, dma_mcux_edma_irq_handler (dev=0x3004c4b0 <__device_dts_ord_486>, channel=1) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:191
191		uint32_t hw_channel = dma_mcux_edma_add_channel_gap(dev, channel);
(gdb) x/16w 0x40071000
0x40071000:	0x80aa6132	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x03010000	0x80000180	0x0372000a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa528c	0x03700002	0x80000200	0x03720012
(gdb) c
Continuing.

Breakpoint 2, dma_mcux_edma_irq_handler (dev=0x3004c4b0 <__device_dts_ord_486>, channel=1) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:195
195			LOG_DBG("IRQ OCCURRED");
(gdb) x/16w 0x40071000
0x40071000:	0x80aa6136	0x01010002	0x00000002	0x00000000
0x40071010:	0x40404020	0x02ff0000	0x80000180	0x0372000a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa5290	0x036e0002	0x80000200	0x03720012
(gdb) c
Continuing.

Breakpoint 1, dma_mcux_edma_reload (dev=0x3004c4b0 <__device_dts_ord_486>, channel=1, src=1077952672, dst=2158651188, size=1764) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:476
476		struct call_back *data = DEV_CHANNEL_DATA(dev, channel);
(gdb) x/16w 0x40071000
0x40071000:	0x80aa616a	0x01010002	0x00000002	0x00000000
0x40071010:	**0x40404020**	0x02e50000	0x80000180	0x0372000a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa52c4	0x03540002	0x80000200	0x03720012
(gdb) c
Continuing.

Breakpoint 2, dma_mcux_edma_irq_handler (dev=0x3004c4b0 <__device_dts_ord_486>, channel=0) at /home/radekp/projects/zephyrproject/zephyr/drivers/dma/dma_mcux_edma.c:191
191		uint32_t hw_channel = dma_mcux_edma_add_channel_gap(dev, channel);
(gdb) x/16w 0x40071000
0x40071000:	0x80aa6734	0x01010002	0x00000002	0x00000000
0x40071010:	**0xc04041a0**	0x03720000	0x80000180	0x0372008a
0x40071020:	0x404040a0	0x01010000	0x00000002	0x00000000
0x40071030:	0x80aa5894	0x006c0002	0x80000200	0x03720012
(gdb) 

After that line CPU is stuck on handling dma_mcux_edma_error_irq_handler

dleach02, 2024-03-28:
@RadekPolyend, thank you for the report. Is there a Zephyr test case or sample program being used to reproduce the problem being reported here?

RadekPolyend, 2024-03-29:
The sample application I use is i2s echo with dts overlays for mimxrt1170_evk of course.

DerekSnell, 2024-04-11:
Hi @RadekPolyend ,
Thank you for reporting this.  I do not yet have a solution for this issue.  But I wanted to point out that the eDMA TCD, which includes the DADDR, can be updated by hardware.  This is a feature of the eDMA.  Which is why a watchpoint at that address may not catch when it is updated, since watchpoints only monitor the address bus from the CPU.

Thanks

RadekPolyend, 2024-04-12:
Hi @DerekSnell ,
Yes that's what I figured out the question is what circumstances may lead to the fact that eDMA puts some invalid address there ? What are the use cases when DESTADDR is changed in this case by eDMA because I couldn't figure it out from the reference manual ? Could you point me to the part of documentation that described that ?

DerekSnell, 2024-04-12:
Hi @RadekPolyend ,
The I2S driver uses the scatter/gather mode of the eDMA to keep a stream of DMA buffers queued up.  With scatter/gather, the eDMA will update the TCD automatically after the major loop completes.  I assume that is when the destination address gets updated with the invalid address.  You can read more about the scatter/gather feature in the eDMA chapter of the Reference Manual.  

Best regards

RadekPolyend, 2024-04-13:
Hi @DerekSnell ,
yeah but this is a peripheral address for TX transfer which should not change at all within this context

RadekPolyend, 2024-06-09:
Recently when I was analyzing IRQs on the board I found out that there is a I2S RX irq handler present in the generated code. Maybe that's an issue ? All in all if DMA is supposed to handle data receive CPU somehow interfere with it casuing this undefined behaviour.

closed 2024-11-08 by mmahadevan108

decsny, 2025-01-21:
@RadekPolyend is this issue resolved ?

commit 7145295a (2024-11-08): Drivers: driver: i2s: Use eDMA loop SG mode

I2S speed test failed because of "Invalid destination address" issue By use
eDMA loop SG mode, this issue is fixed.

Fixes: #70777