zephyrproject-rtos/zephyr #70778: Error in TX callback of grow_r502a sensor driver

flabou, 2024-03-27:
I think there is a mistake in this grow_r502a sensor driver, specifically at the code snippet referenced there:

https://github.com/zephyrproject-rtos/zephyr/blame/dd0923d7ad8ac1bac39bcbb55418ebb678591991/drivers/sensor/grow_r502a/grow_r502a.c#L60-L64

```C
	while (drv_data->tx_buf.len) {
		sent = uart_fifo_fill(config->dev, &drv_data->tx_buf.data[sent],
							drv_data->tx_buf.len);
		drv_data->tx_buf.len -= sent;
	}
```
I believe that in that way, the same bytes will be sent repeatedly unless the FIFO is big enough to send all in two calls.

I have not tested this with this driver directly as I don't have this sensor, but I used this driver as a reference for another UART driver as an exercise, and I realized the frame were not sent properly, in my case I fixed it like so:

```C
	while(sent < drv_data->tx_buf.len){
		sent += uart_fifo_fill(config->dev, &drv_data->tx_buf.data[sent],
							drv_data->tx_buf.len);
	}
	drv_data->tx_buf.len = 0;
```

I hope I am not mistaken and this is helpful.

nordicjm, 2024-03-28:
@DineshDK03

henrikbrixandersen, 2024-03-28:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

DineshDK03, 2024-04-01:
https://github.com/zephyrproject-rtos/zephyr/pull/59654#event-12223848706 . This issue addressed in this PR.

MaureenHelm, 2024-06-03:
> [#59654 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/59654#event-12223848706) . This issue addressed in this PR.

Fixed by #59654

closed 2024-06-03 by MaureenHelm