zephyrproject-rtos/zephyr #70792: Wi-Fi: Compilation Errors

WilliamGFish, 2024-03-27:
**Description**
The changes made to the Wi-fi shell it fails to compile(Add support to configure BSSID in Wi-Fi connect command #70024). This is a known issue with C case statements.

> /subsys/net/l2/wifi/wifi_shell.c:507:4: error: a label can only be part of a statement and a declaration is not a statement
>   507 |    long channel = strtol(optarg, &endptr, 10);

**Impact**
Showstopper

**Logs and console output**

> /subsys/net/l2/wifi/wifi_shell.c:507:4: error: a label can only be part of a statement and a declaration is not a statement
>   507 |    long channel = strtol(optarg, &endptr, 10);

**Additional context**
By adding an additional statement the compiler ignores that 'issue'
```
		case 'c': //Break Comment to resolve compile error
			long channel = strtol(optarg, &endptr, 10);
```

jukkar, 2024-03-27:
What compiler is that? I do not see any error with gcc.

WilliamGFish, 2024-03-27:
Using GCC...

> 
> -- west build: generating a build system
> Loading Zephyr default modules (Zephyr base).
> -- Application: C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server
> -- CMake version: 3.25.0-rc2
> -- Found Python3: C:\Python310\python.exe (found suitable version "3.10.8", minimum required is "3.8") found components: Interpreter
> -- Cache files will be written to: C:\Users\BillyBob\AppData\Local/.cache/zephyr
> -- Zephyr version: 3.6.99 (C:/Users/BillyBob/zephyrGoodX/zephyrNew)
> -- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
> -- Board: particle_argon, qualifiers: nrf52840
> -- Found toolchain: gnuarmemb (C:\gnu_arm_embedded_10)
> -- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
> -- Found BOARD.dts: C:/Users/BillyBob/zephyrGoodX/zephyrNew/boards/particle/argon/particle_argon.dts
> -- Found devicetree overlay: C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/boards/particle_argon.overlay
> -- Generated zephyr.dts: C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/build/zephyr/zephyr.dts
> -- Generated devicetree_generated.h: C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/build/zephyr/include/generated/devicetree_generated.h
> -- Including generated dts.cmake file: C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/build/zephyr/dts.cmake
> Parsing C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/Kconfig
> Loaded configuration 'C:/Users/BillyBob/zephyrGoodX/zephyrNew/boards/particle/argon/particle_argon_defconfig'
> Merged configuration 'C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/prj.conf'
> Merged configuration 'C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/boards/particle_argon.conf'
> Merged configuration 'C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/overlay-local-mqtt.conf'
> Configuration saved to 'C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/build/zephyr/.config'
> Kconfig header saved to 'C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server/build/zephyr/include/generated/autoconf.h'
> -- Found GnuLd: c:/gnu_arm_embedded_10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe (found version "2.36.1")
> -- The C compiler identification is GNU 10.3.1
> -- The CXX compiler identification is GNU 10.3.1
> -- The ASM compiler identification is GNU
> -- Found assembler: C:/gnu_arm_embedded_10/bin/arm-none-eabi-gcc.exe

krish2718, 2024-03-27:
Are you doing some extra configuration to run gcc in `-pedantic` mode? From https://stackoverflow.com/a/77738076  looks like it would work with gcc otherwise (though it shouldn't work).

PR #70796 (2024-03-27): net: wifi: shell: Avoid declaring variables after case statement
Some compilers have trouble parsing variables directly after a case statement.

Fixes: #70792
commit 42356b6d (2024-03-27): net: wifi: shell: Avoid declaring variables after case statement

Some compilers have trouble parsing variables directly after a case
statement.

Fixes: #70792
closed 2024-03-27 by dleach02