zephyrproject-rtos/zephyr #70794: Bluetooth isochronous channels central_iso / peripheral_iso example not working reliable on nrf52840dk – packets not received/lost

Benni77, 2024-03-27:
**Describe the bug**
I tried the two examples: [CENTRAL_ISO](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/central_iso) & [PERIPHERAL_ISO](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/peripheral_iso)
with two nrf52840, one running central and one running peripheral.

Problem:
Data stream on iso channel stops at some point.

Software flow:
1. Connection is established between central and peripheral.
2. ISO channel is connected succesfully.
3. Central starts data stream (all parameters from example left unchanged)
4. Peripheral receives some data and prints messages to console.
5. At some point (after few intervals) the peripheral is not receiving any more data, while central is still "sending".

I can extend data stream duration by increasing the channel interval.

 - What target platform are you using?
 nRF52840

 - What have you tried to diagnose or workaround this issue?
I checked with Segger's Systemview and saw that the BT TX thread is called periodically on the central, but the BT RX thread is not called on the peripheral after data reception has stopped.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash CENTRAL_ISO and PERIPHERAL_ISO to two seperate nrf52840DK Boards
2. Let them execute the app and observe received data on peripheral
3. See that no more data arrives after a certain point in time

**Expected behavior**
Reliable data stream on the Bluetooth LE Isochronous channel between the two devices.

**Impact**
As it works using the nRF Connect SDK 2.6.0 (SoftDevice Controller), this can be used as an alternative.

**Logs and console output**
**CENTRAL**
```
*** Booting Zephyr OS build v3.6.0-1517-g38ad4b6d43cf ***
[00:00:00.288,330] [0m<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)[0m
[00:00:00.288,360] [0m<inf> bt_hci_core: HW Variant: nRF52x (0x0002)[0m
[00:00:00.288,391] [0m<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.6 Build 99[0m
[00:00:00.289,093] [0m<inf> bt_hci_core: Identity: C1:E4:D9:06:37:ED (random)[0m
[00:00:00.289,123] [0m<inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1[0m
[00:00:00.289,154] [0m<inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff[0m
Bluetooth initialized
Scanning successfully started
Device found: C4:C7:C1:86:A3:3B (random) (RSSI -39)
Connected: C4:C7:C1:86:A3:3B (random)
ISO Channel 0x20001d00 connected
```

**PERIPHERAL**
```
00> *** Booting Zephyr OS build v3.6.0-1517-g38ad4b6d43cf ***
00> [00:00:00.479,125] <inf> fs_nvs: 8 Sectors of 4096 bytes
00> [00:00:00.479,125] <inf> fs_nvs: alloc wra: 0, fd0
00> [00:00:00.479,125] <inf> fs_nvs: data wra: 0, 1c
00> [00:00:00.481,109] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
00> [00:00:00.481,140] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
00> [00:00:00.481,170] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.6 Build 99
00> [00:00:00.481,567] <inf> bt_hci_core: No ID address. App must call settings_load()
00> [00:00:00.481,964] <inf> bt_hci_core: Identity: C4:C7:C1:86:A3:3B (random)
00> [00:00:00.481,994] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
00> [00:00:00.482,025] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
00> Bluetooth initialized
00> Advertising successfully started
00> added message from main
00> added message from main
00> added message from main
00> Connected C1:E4:D9:06:37:ED (random)
00> Incoming request from 0x20000eb8
00> ISO Channel 0x20000190 connected
00> Incoming data channel 0x20000190 len 1
00>    00
00> Incoming data channel 0x20000190 len 2
00>    0001
00> Incoming data channel 0x20000190 len 3
00>    000102
00> Incoming data channel 0x20000190 len 4
00>    00010203
00> Incoming data channel 0x20000190 len 5
00>    0001020304
00> Incoming data channel 0x20000190 len 6
00>    000102030405
00> Incoming data channel 0x20000190 len 7
00>    000102...040506

...

00>    000102...565758
00> Incoming data channel 0x20000190 len 90
00>    000102...575859
00> Incoming data channel 0x20000190 len 91
00>    000102...58595a
00> Incoming data channel 0x20000190 len 92
00>    000102...595a5b
00> added message from main
00> added message from main
00> added message from main
00> added message from main
00> added message from main
00> added message from main
```

**Environment (please complete the following information):**
 - OS: Manjaro – Linux
 - zephyr-sdk-0.16.5-1
 - Zephyr 3.6.99

cvinayak, 2024-03-29:
@Benni77 Please try with the below added in the central_iso's prj.conf

```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y
```

Benni77, 2024-03-29:
> @Benni77 Please try with the below added in the central_iso's prj.conf
> 
> 
> 
> ```
> 
> CONFIG_BT_CTLR_ADVANCED_FEATURES=y
> 
> CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y
> 
> ```

Thanks for the answer! I will give it a try on Tuesday when holidays are over and I'm at work. 
I'll let you know the result.

Benni77, 2024-04-02:
Hi @cvinayak,

I tried with the two configs added on central_iso. And it seems to work, as long as sent buf array is shorter than 247 (HCI ISO data pack size?). If buf array length > 247 it again stops streaming.

([main.c; L49](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/central_iso/src/main.c#L49))

`static uint8_t buf_data[CONFIG_BT_ISO_TX_MTU];` _stops streaming after message length 247_
->
`static uint8_t buf_data[247];` _seems to work_

Here are the logs on peripheral side:

```
00> *** Booting Zephyr OS build v3.6.0-1517-g38ad4b6d43cf ***
00> [00:00:00.259,277] <inf> fs_nvs: 8 Sectors of 4096 bytes
00> [00:00:00.259,277] <inf> fs_nvs: alloc wra: 0, fd0
00> [00:00:00.259,277] <inf> fs_nvs: data wra: 0, 1c
00> [00:00:00.261,291] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
00> [00:00:00.261,291] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
00> [00:00:00.261,322] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.6 Build 99
00> [00:00:00.261,718] <inf> bt_hci_core: No ID address. App must call settings_load()
00> [00:00:00.262,115] <inf> bt_hci_core: Identity: C4:C7:C1:86:A3:3B (random)
00> [00:00:00.262,145] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
00> [00:00:00.262,176] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
00> Bluetooth initialized
00> Advertising successfully started
00> Connected E1:51:F8:42:2D:51 (random)
00> Incoming request from 0x20000ec0
00> ISO Channel 0x20000190 connected
00> Incoming data channel 0x20000190 len 1
00>    00
00> Incoming data channel 0x20000190 len 2
00>    0001

...

00>    000102...f2f3f4
00> Incoming data channel 0x20000190 len 246
00>    000102...f3f4f5
00> Incoming data channel 0x20000190 len 247
00>    000102...f4f5f6
```

cvinayak, 2024-04-02:
This should help:
```
CONFIG_BT_ISO_TX_MTU=247
CONFIG_BT_ISO_RX_MTU=247
CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE=255

CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y
```

I will investigate further if `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE=255`  is the right value to be used.

cvinayak, 2024-05-02:
Some samples have the below comment, but maybe the right thing is to have the increased bytes in the Controller's CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` as done in this sample:
https://github.com/zephyrproject-rtos/zephyr/blob/ea995f6378f7f6a6f8b6b1f7df29e0af7437f7a1/samples/bluetooth/broadcast_audio_source/overlay-bt_ll_sw_split.conf#L17-L24

Host or Controller does not need any fix, the comment about fixing host should be removed? @Thalley ?

Thalley, 2024-05-02:
> Some samples have the below comment, but maybe the right thing is to have the increased bytes in the Controller's CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` as done in this sample: [...]

As discussed previously I do think something needs to change so that `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` can default to `CONFIG_BT_ISO_TX_MTU` (or even just use `CONFIG_BT_ISO_TX_MTU` in the controller). In my mind that makes the most sense.

rugeGerritsen, 2024-05-02:
> > Some samples have the below comment, but maybe the right thing is to have the increased bytes in the Controller's CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` as done in this sample:
> > https://github.com/zephyrproject-rtos/zephyr/blob/ea995f6378f7f6a6f8b6b1f7df29e0af7437f7a1/samples/bluetooth/broadcast_audio_source/overlay-bt_ll_sw_split.conf#L17-L24
> > 
> > Host or Controller does not need any fix, the comment about fixing host should be removed? @Thalley ?
> 
> As discussed previously I do think something needs to change so that `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` can default to `CONFIG_BT_ISO_TX_MTU` (or even just use `CONFIG_BT_ISO_TX_MTU` in the controller). In my mind that makes the most sense.
> 
> @rugeGerritsen You may also have some input here

This sounds like a good idea. The SoftDevice controller uses the MTU size for configuring memory buffer sizes: https://github.com/nrfconnect/sdk-nrf/blob/main/subsys/bluetooth/controller/hci_driver.c#L1104

Note: `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` needs to be set to `CONFIG_BT_ISO_TX_MTU + 8` if you want to support sending HCI ISO packets with timestamp to the controller. The `8` corresponds to the ISO SDU header and timestamp size.

PR #72354 (2024-05-10): Bluetooth: Controller: BT_CTLR_ISO_TX_BUFFER_SIZE from BT_ISO_TX_MTU
Derive BT_CTLR_ISO_TX_BUFFER_SIZE from BT_ISO_TX_MTU when BT_ISO is enabled.

Fixes #70794.
closed 2024-05-10 by carlescufi