zephyrproject-rtos/zephyr #70841: gpio_basic_api has errors when setting GPIO_INT_EDGE_BOTH

RichardSWheatley, 2024-03-28:
**Describe the bug**
I beleive a feature that is not supported should not fail any test.

In tests\drivers\gpio\gpio_basic_api\test_config_trigger.c:
- If GPIO_INT_EDGE_BOTH is -ENOTSUP, the zassert_ok will return a failure.

Since this feature may not be supported by every soc, the test should not fail but simply skip that part of the test.
The Ambiq apollo4 plus series has an errata that prohibits the use of this feature.

**To Reproduce**
Build the test with an overlay using the following for the apollo4p_evb:
`&counter0 {
    status = "okay";
};`

The following code will show a failure:
`ret = gpio_pin_interrupt_configure(dev, PIN_IN, GPIO_INT_EDGE_BOTH);`
`zassert_ok(ret, "enable callback failed");`

A simple work-around would be:
`ret = gpio_pin_interrupt_configure(dev, PIN_IN, GPIO_INT_EDGE_BOTH);`
`if (ret == -ENOTSUP) {`
`    TC_PRINT(" * GPIO_INT_EDGE_BOTH not supported.\n");`
`} else {`
`    zassert_ok(ret, "enable callback failed");`
`}`

**Expected behavior**
The tests will fail instead of showing the ENOTSUP.

**Impact**
This is not a showstopper but correct procedure should be followed.
Other ENOTSUP features are supported in this test.

**Logs and console output**
As-is test code returns the following results:

> START - test_gpio_config_trigger
> 
> Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_config_trigger.c:90: after_flash_gpio_config_trigger_test_gpio_config_trigger: (ret is non-zero)
> enable callback failed
>  FAIL - test_gpio_config_trigger in 0.019 seconds
> 
> START - test_gpio_config_twice_trigger
> 
>Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_config_trigger.c:43: after_flash_gpio_config_trigger_test_gpio_config_twice_trigger: (ret is non-zero)
> enable callback failed
>  FAIL - test_gpio_config_twice_trigger in 0.019 seconds

After workaround the output is:

> START - test_gpio_config_trigger
> - GPIO_INT_EDGE_BOTH not supported.
> PASS - test_gpio_config_trigger in 0.016 seconds
> 
> START - test_gpio_config_twice_trigger
> - GPIO_INT_EDGE_BOTH not supported.
> PASS - test_gpio_config_twice_trigger in 0.015 seconds

**Environment (please complete the following information):**
main Zephyr branch

**Additional context**
None

henrikbrixandersen, 2024-04-04:
The test should be skipped if `GPIO_INT_EDGE_BOTH` is not supported. Are you able to send a PR for updating the test?

tbr-tt, 2024-04-04:
The PR #65816 should fix this issue.

RichardSWheatley, 2024-04-04:
> The PR #65816 should fix this issue.

I do not think it would skip the subtest but would skip the rest of that test.

That test has 5 subtests and you just skipped the remaining 3.

That Z_TEST
`ZTEST(after_flash_gpio_config_trigger, test_gpio_config_twice_trigger)`
...
`	/* 2. Enable PIN callback as both edges */`
`	ret = gpio_pin_interrupt_configure(dev, PIN_IN, GPIO_INT_EDGE_BOTH);`
`	if (ret == -ENOTSUP) {`
`		TC_PRINT("Both edge GPIO interrupt not supported.\n");`
`		gpio_remove_callback(dev, &drv_data->gpio_cb);`
`		ztest_test_skip();`
`		return;`
`	}`
`	zassert_ok(ret, "enable callback failed");`
...

My proposal is this:
`	/* 2. Enable PIN callback as both edges */`
`	ret = gpio_pin_interrupt_configure(dev, PIN_IN, GPIO_INT_EDGE_BOTH);`
`	if (ret == -ENOTSUP) {`
`		TC_PRINT("Both edge GPIO interrupt not supported.\n");`
`		gpio_remove_callback(dev, &drv_data->gpio_cb);`
`	}`
`	else {`
`		zassert_ok(ret, "enable callback failed");`
`	}`

RichardSWheatley, 2024-04-04:
Here are the test results with my changes.

> ------ TESTSUITE SUMMARY START ------
> 
> SUITE PASS - 100.00% [after_flash_gpio_config_trigger]: pass = 2, fail = 0, skip = 0, total = 2 duration = 0.032 seconds
>  - PASS - [after_flash_gpio_config_trigger.test_gpio_config_trigger] duration = 0.016 seconds
>  - PASS - [after_flash_gpio_config_trigger.test_gpio_config_twice_trigger] duration = 0.016 seconds
> 
> SUITE PASS - 100.00% [gpio_port]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.023 seconds
>  - PASS - [gpio_port.test_gpio_port] duration = 0.023 seconds
> 
> SUITE PASS - 100.00% [gpio_port_cb_mgmt]: pass = 3, fail = 0, skip = 0, total = 3 duration = 9.745 seconds
>  - PASS - [gpio_port_cb_mgmt.test_gpio_callback_add_remove] duration = 3.615 seconds
>  - PASS - [gpio_port_cb_mgmt.test_gpio_callback_enable_disable] duration = 3.616 seconds
>  - PASS - [gpio_port_cb_mgmt.test_gpio_callback_self_remove] duration = 2.514 seconds
> 
> SUITE PASS - 100.00% [gpio_port_cb_vari]: pass = 1, fail = 0, skip = 0, total = 1 duration = 4.427 seconds
>  - PASS - [gpio_port_cb_vari.test_gpio_callback_variants] duration = 4.427 seconds
> 
> ------ TESTSUITE SUMMARY END ------

RichardSWheatley, 2024-04-04:
https://github.com/zephyrproject-rtos/zephyr/pull/71108

closed 2024-04-20 by RichardSWheatley