zephyrproject-rtos/zephyr #70858: Arduino GIGA Display Shield support

maarten1C96, 2024-03-28:
I am new to Zephyr and happy to see the Arduino GIGA R1 is being supported. After installing the basics I have the blink sample successfully running now. I also have an Arduino GIGA Display Shield connected to this board, which I would like to use with LVGL. Is it correct to think this shield is not yet being supported by Zephyr? And if not, are there any future plans for doing so?

maarten1C96, 2024-04-27:
And given the capabilities of other development kits that include a display connected via **MIPI DSI**, like the `STM32H747I-DISCO` from STMicroelectronics, the `i.MX RT1050` from NXP, and others, would it not be highly beneficial for these brands to provide examples on how to communicate video data using this DSI protocol within Zephyr? 🥰

facchinm, 2025-02-04:
Hi @maarten1C96 , we are working on it, stay tuned 🙂

mariopaja, 2025-04-07:
I guess this can be closed now? https://github.com/zephyrproject-rtos/zephyr/pull/85569

closed 2025-04-07 by facchinm